canopen-python/canopen #222: Receiving and Transmitting different node PDOs concurrently

edesta-be, 2021-02-12:
Hello!

So I have been playing around with this library and had a few questions.

I have a system that will have up to 6 motors and wanted to be able to send and receive velocity commands to each of them respectively.

My current setup is currently just two motors because i was trying to determine feasibility of what I am trying to achieve. 
So in short, I have mapped the PDOs for both the nodes to transmit on information updates. I have added a callback for the nodes tpdo to just print the current velocities as it changes. (For more background, I am commanding velocities in a separate thread)

This works fine with a single node. My confusion comes from wanting to have multiple node. When i add a secondary node, i go through the same motions of mapping the respective pdo and ensuring everything is fine,  in the same thread that I use to update the RPDO value of the first node, i update the RPDO of the secondary node. But after one message, its almost as if the bus dies (idk how else to explain it). It wont update the velocities again, and it requires a full system power cycle for the motors to respond again. 
When this happens also, the TPDO callback no longer occurs.

Is there something I am missing? I feel like I might just be doing something foolish.

Thank you for your time!
Look forward to your response!

edesta-be, 2021-02-12:
Here is my code! Figured I would lay it out in case I'm doing something dumb. 
I also found an issue that sounded similar with using the python-can library thread safe bus? Is that perhaps what I am missing, if so how do i go about making the change.

Thanks again!!!
```

import canopen
import os
import time
import threading

n = 6000
m = 6000
EDS = os.path.join('samples', 'subsys', 'canbus', 'canopen', 'objdict', 'objdict.eds')

def node25_commandvelocity(node):
    global n
    while True:
        node.rpdo[3][0x6040].raw = 15 # setting bit 4 and 5 to 0
        node.rpdo[3][0x60FF].raw = n
        node.rpdo[3].transmit()

        time.sleep(3)

        n = int(n-1000)
        print(f'n: {n}')
        if n <= 100:
            break

def node26_commandvelocity(node):
    global m
    while True:
        node.rpdo[3][0x6040].raw = 15 # setting bit 4 and 5 to 0
        node.rpdo[3][0x60FF].raw = m
        node.rpdo[3].transmit()

        time.sleep(10)

        m = int(m*0.65)
        print(f'm: {m}')
        if m <= 100:
            break

def print_speed(message):
    print('%s received' % message.name)
    for var in message:
        print('%s = %d' % (var.name, var.raw))

def init_pdos(node):
    node.tpdo.read()
    node.rpdo.read()

    # Re-map PDO[3]
    node.tpdo[3].clear()
    node.tpdo[3].add_variable(0x6041)
    node.tpdo[3].add_variable(0x606C)
    node.tpdo[3].trans_type = 255
    node.tpdo[3].enabled = True
    node.rpdo[3].clear()
    node.rpdo[3].add_variable(0x6040)
    node.rpdo[3].add_variable(0x60FF)
    node.rpdo[3].enabled = True

    node.tpdo.save()
    node.rpdo.save()

    node.nmt.state = 'OPERATIONAL' # put the controller in operational
    node.sdo[0x6060].raw = 3
    time.sleep(2)

    node.rpdo[3][0x6040].raw = 6 # ready to switch on
    node.rpdo[3].transmit()
    time.sleep(1)

    node.rpdo[3][0x6040].raw = 7 # ready to switch on
    node.rpdo[3].transmit()
    time.sleep(1)

    node.rpdo[3][0x6040].raw = 271 # operation enabled state with motion halted
    node.rpdo[3].transmit()
    time.sleep(1)

def init_motors():

    network = canopen.Network()
    network.connect()

    node = canopen.BaseNode402(25, EDS)
    network.add_node(node)

    node2 = canopen.BaseNode402(26, EDS)
    network.add_node(node2)

    node.nmt.state = 'PRE-OPERATIONAL' # set controller into pre operational
    node2.nmt.state = 'PRE-OPERATIONAL' # set controller into pre operational

    init_pdos(node)
    time.sleep(1)
    init_pdos(node2)

    t2 = threading.Thread(target=node25_commandvelocity, args=(node,))
    t2.start()

    t3 = threading.Thread(target=node26_commandvelocity, args=(node2,))
    t3.start()

    node.tpdo[3].add_callback(print_speed)
    node2.tpdo[3].add_callback(print_speed)

    global n
    global m
    while True:
        if n <= 100 and m <= 100:
            break

    t2.join()
    t3.join()

    node.rpdo[3][0x6040].raw = 271 # operation enabled state with motion halted
    node.rpdo[3].transmit()
    node2.rpdo[3][0x6040].raw = 271 # operation enabled state with motion halted
    node2.rpdo[3].transmit()
    network.disconnect()

if __name__ == "__main__":
    init_motors()
    print("Done!")
```

af-silva, 2021-02-13:
Hi @edesta-be ,

`I also found an issue that sounded similar with using the python-can library thread safe bus?`

Most likely, I'm not sure if python-can provides a thread save mechanism to access the can BUS aka configured device. So you can be having problems because the two threads are trying to capture the access to de device. 

I never face this problem since my implementation base on the DS402 base nod uses ROS as backend and I have only one thread accessing the can network, and I capture the message from the nodes and use the pub/subscribe mechanism. 
I think you can explore a python framework that implements pub/sub our event driven like circuits for maintaining a main thread accessing the bus and then capturing calls for send and receive data from the can network individualy for each node.

In my use case I have 8 motors moving simultaneously but synchronize the movement and use the stop guard to halt movement in case of communication lost with the Master node (application running on the machine). Using this approach, a read all the values for each node and set the TPDOS to the correct values, then at sync #80 all the motors move to the desire position/velocity/command.

I'm replying from my phone, and hope this helps. I will try to check better when I have access to a computer ... sorry, on the move right now.

Best regards.

edesta-be, 2021-02-14:
Oh sweet I was planning to use ROS2 as the backend so this might actually be worth me transitioning it over. 
Okay cool just so I understand fully as well. You used SYNC messages to transmit and receive messages instead of only just transmitting or receiving on changes? 
and with a ROS implementation is it worth adding a callback on TPDO event?

af-silva, 2021-02-15:
Hi,

>You used SYNC messages to transmit and receive messages instead of only just transmitting or receiving on changes?

The SYNC message `#80` serves to sync the motor actuation, think of velocity motors, i want each motor to move at the same time, even if at different velocities. 

``` python
... 
while not rospy.is_shutdown():
  # updates the diagnostic aggregator
  self.updater.update()
  # check if network is still ok (this troughs a exception)
  self.network.check()
  # check last time a command was received 
  self._check_last_command()
  # give some rest to the oompa loompas
  loop_rate.sleep()

...
```

Then I have my subscriber to receive the velocities for each motor and the publisher specific for each node do pub the status of each motor, also I use a diagnostic aggregator to collect information (e.g., temperature, up time, erros ...).

```Python 
def _sub_control_commands(self, msg):

 #... <other code here>

 for node in self.vnodes.values():                
    if node.state != 'OPERATION ENABLED':
        break                  
    if node.name == 'left_motor':                        
        _left = node                        
    elif node.name == 'right_motor':
        _right = node

# ... <other code here>
```

>and with a ROS implementation is it worth adding a callback on TPDO event?

The callbacks are set using the ROS pub/sub mechanism. I use the TPDO/RPDO to send/receive messages from the nodes, the only callback set is the emcy on the NMT object, that notifies if something went wrong with one of the nodes. The messages to and from the motors are write/read at a specific interval defined by the pubrate of the ROS node. 
Basically, I wrapped the  `BaseNode402` object for each type of motor (velocity, position) and set up a ROS node to control each of the motors through Pub/Sub messages.

I hope this helps.

edesta-be, 2021-02-18:
Amazing thank you! I will try this out!
Question? I have managed to SYNC transmission from the motor nodes however, how do I also transmit from my node to the Motor nodes on SYNC?
```
node.tpdo[3].clear()
node.tpdo[3].add_variable(0x6041)
node.tpdo[3].add_variable(0x606C)
node.tpdo[3].trans_type = 1
node.tpdo[3].enabled = True
node.rpdo[3].clear()
node.rpdo[3].add_variable(0x6040)
node.rpdo[3].add_variable(0x60FF)
node.rpdo[3].enabled = True
```

I tried setting the trans_type of the rpdo as 1. I tried to follow starting RPDO transmission on a selected rate (however, that still caused the above error, is there a way to send RPDOs on SYNC or am I mistaken?

edesta-be, 2021-02-18:
So i kept at it:
I changed the trans_type of the rpdo to 1 for both motors I am experimenting with. 
Used a callback to call per SYNC message. 

And when I call the motor commands (pretty much transmit the first one no delay transmit the second one) -- it happens as it was happening prior with the motors hearing the first command and just ignoring the other ones coming in, even though i still see the commands being sent over. 

It might have to do with the controller as I am using a dual axis controller to control these two motors, so I am checking in with the guys who work on the controller specifically, but wanted to know your thoughts @af-silva  on sending messages back to back
```
node.rpdo[3][0x6040].raw = 15 # setting bit 4 and 5 to 0
node.rpdo[3][0x60FF].raw = n
node.rpdo[3].transmit()
node2.rpdo[3][0x6040].raw = 15 # setting bit 4 and 5 to 0
node2.rpdo[3][0x60FF].raw = n
node2.rpdo[3].transmit()
```

af-silva, 2021-02-19:
Hi,

I may have mislead you, sorry.

>I have managed to SYNC transmission from the motor nodes however, how do I also transmit from my node to the Motor nodes on SYNC?

The SYNC message has I mention is the CAN mechanism that enables the remote controllers to actuate at the same time... when sending commands from the Master node (e.g., your computer) you set the RPDOs with the desire values and then call the `transmit()` function, this will send the message through the CAN network to the remote device when called but the motion will only happen in sync (if sync is configured). So, let's assume that the sync period is 500ms and that you send two consecutive messages separated from 100ms, this means that the two messages will arrive at the remote device before sync, only the last one will take effect. The sync message is configured at the master by calling the `network.sync.start(0.01) # transmit every 100ms` that will periodically send the `#80` sync to the controllers. Other things can be configured on the master, like the node guard (heart beat mechanism) that guarantees that if one node fails (i.e., in this case the master itself) all the other nodes on the network halt their operation.

>It might have to do with the controller as I am using a dual axis controller to control these two motors

Most likely, double check the configuration on the controller and check how it handles the two different actuators, if they are seen as two different devices or as one. If they are two different nodes in the CAN network check the configuration loaded in each node.

The controller shutdown always after it receives the first message??

edesta-be, 2021-02-19:
Yes! Perfect, that was how i was handling SYNC so its good to have my sanity checked there!

In terms of the motor controller, its actually infuriating. I had a second motor controller lying around and wired up the motors through that so I had two controllers and two motors, and I was able to send out the simultaneous commands without any problems. 

I'm using a Dual Axis Copley Controller -- and they register as different nodes, but I'm not sure why I cant command two motors to move via RPDOs OR SDOs which is really strange. 

I thought it might have been a memory address thing with two motors sharing the same controller, so i went ahead and changed the COB-ID. Still no luck.

And actually it fails on the first message sent to the second motor. Like I cant even understand what could be going on. 

And no problem @af-silva ! Haha your help has been awesome

edesta-be, 2021-02-19:
@af-silva A follow up question -- is it possible to send an RPDO to two nodes if I map the COB-ID of both nodes for the RPDO to equate one another.

For instance if i set the COB-ID of RPDO3 to 0x0500 for both Axis A and Axis B. How can I use this stack to send one RPDO to both. Because currently i specifically have to call transmit on the node that i am using: node.rpdo[3].transmit().

af-silva, 2021-02-19:
Each node in the CAN network must have a unique identifier, COB-ID, so you must configure the RPDO for each of the nodes and execute the transmit for each, e.g., node1 and node2, this is done by design. Each node should had its one configuration, that map the desire objects you need to control the remote devices.

>Because currently i specifically have to call transmit on the node that i am using: node.rpdo[3].transmit().

If you have many remote devices, store all the nodes in a list and iterate the list setting the desire objects (e.g., target velocity) and call transmit.

edesta-be, 2021-02-19:
@af-silva using your mapping example:

```
   node2.rpdo[3].clear()
    node2.rpdo[3].add_variable(0x6040)
    node2.rpdo[3].add_variable(0x60FF)
    node2.rpdo[3].enabled = True
    node2.rpdo[3].trans_type = 1
```
Would I configure the COB-ID here if so how? Or should I use an SDO (which is what I am currently doing and that has not proven to be successful). 

Because when i use the above example, it also updates the number of mapped items etc.

acolomb, 2021-02-19:
> @af-silva A follow up question -- is it possible to send an RPDO to two nodes if I map the COB-ID of both nodes for the RPDO to equate one another.

That is definitely possible, I've done it with this library.  You can reconfigure the PDO's COB ID, which will be saved to an OD object via SDO when you call `pdo.save()`.  That is, **IF** the controller supports such re-mapping.  I've dealt with some that would acknowledge the configuration SDO, but not really apply it.  When setting it through their internal mini programming language, the COB-ID could be changed though.  If you send that RPDO from your Python script then, both controllers will react to it and pull out the values according to their respective configuration.

This is not to be confused with the Node-ID.  Be *default*, the PDOs have standard COB-IDs based on the Node-ID.  That is only the so called "pre-defined connection set" however, and the COB-IDs for each PDO and SDO can be reassigned at will.  That is a very wise design, because several devices can be instructed with the same PDO simultaneously, with less bus load.  Note that this doesn't apply to SDOs, because they must be confirmed by the recipient.

af-silva, 2021-02-19:
Ok, I stand corrected. I never used it in such way, but yes the cob-id can be configured for each PDO by the configuration loaded (eds file), I was not seeing this way.

For instance, RPDO 1 configuration -- object 1400, sub index 1 holds the COB-id for which the RPDO 1 responds to.
The default is the Node ID + 0x200, so you could force like @acolomb said, for it to respond to a different cobid, let's say one cobid for both controllers.
(The mapping for RPDO 1 is found at object 1600)

acolomb, 2021-02-19:
It's even simpler by just setting the `.cob_id` property on a specific PDO in python-canopen, before calling `.save()`: https://canopen.readthedocs.io/en/latest/pdo.html#canopen.pdo.Map.cob_id

edesta-be, 2021-02-25:
Amazing! Thank you everyone! This has been amazingly helpful! I have successfully integrated the motors as intended. 

Thank you @af-silva and @acolomb  for your time and answers

For future as well it looks like for anyone using Copley Controllers and this library, if you ever want to move to motors simultaneously, there is a problem with their newest firmware that doesnt allow that for some weird reason. I downgraded to an older version and had the system up and running in no time!

closed 2021-02-25 by edesta-be