canopen-python/canopen #223: Problems receiving bootup message

Neokimx, 2021-02-15:
Hello guys,

I have a Problem receiving the bootup message from my CANopen motor.
Maybe someone can help me to fix this, since I'm not familiar with CANbus and canopen.

My sample code:

`myNetwork = canopen.Network()`
`myNetwork.connect(bustype='ixxat', channel=0, bitrate=500000)`

`myNetwork.nmt.state = "RESET"`
`myNetwork.nmt.wait_for_bootup(timeout=10)`

I can see that the motor resets.
I'm also able to receive the bootup message 0x00 with COB-ID 711 in my IXXAT CAN Software.

But within python I'm never receiving this message and the wait for bootup method raises and error.

Hopefully someone can give me a hint on how to solve this.
Thanks and greetings,
Neokimx

christiansandberg, 2021-02-15:
You need to use the node object instead of the network object when waiting for boot up since the message comes from a particular node id.

Neokimx, 2021-02-15:
Ok, I see. 
Then I do have a follow up question. Maybe you could help me there aswell :)

You suggest to use something like this:

`myNetwork = canopen.Network()`
`myNetwork.connect(bustype='ixxat', channel=0, bitrate=500000)`

`myNetwork.nmt.state = "RESET"`
`myNode.nmt.wait_for_bootup(timeout=10)`

But then I don't know how to get the `myNode` object.
Right know I'm using the `scanner` attribute to scan the network for nodes. 
Thats works and my motor is found.

But then the `myNetwork` object has no node objects.
I might to add that node manually. but without an eds file I don't know how to do that.
In your given examples you add a node manually via eds file, which I don't have.

Might be a noob question. Sorry for that.

christiansandberg, 2021-02-15:
You might be able to use the `network.add_node(node_id)` without supplying an EDS file in this case.

Neokimx, 2021-02-15:
Ok, I will try that tomorrow.

I just realised I will need the myNode Object for SDO request, too.
So I hope this will work.

Thanks so far!

closed 2021-02-15 by Neokimx