canopen-python/canopen #226: Devices not responding to LSS fastscan

edcloudcycle, 2021-02-20:
Hi @ymkim92 and all,

Thanks for the great work on this library, LSS is a really needed feature. I am having trouble getting it to work with my setup. With a single [Hydac HPT1400](https://www.hydac.com/fileadmin/pdb/pdf/PRO0000000000000000000018136000011.pdf) on the bus I can read the SDOs configure the TPDO and set it to send me data no problem all with the default node_id of 1.
I need to have 2 of these devices on the same bus and I was hoping to use LSS to configure them but fastscan returns no results. My LSS code looks like:

`
network = canopen.Network()
network.connect(bustype='socketcan', channel='can1', bitrate=250000)
found_device, lss_id = network.lss.fast_scan()
if found_device == True:
    print('LSS {0} found, reports node id: {1}'.format(lss_id[3], network.lss.inquire_node_id()))
    network.lss.configure_node_id(12)
    network.lss.store_configuration()`

But found_device never reports true. The datasheet does indicate that LSS is supported.

All suggestions very welcome

Thanks

Ed
[HE-928312-0001.pdf](https://github.com/christiansandberg/canopen/files/6014680/HE-928312-0001.pdf)

ymkim92, 2021-12-15:
One thing I suspect is that the device is not in "unconfigured" state.

```
non-configured LSS slave
LSS slave device that owns only an inactive persistent node-ID equal to FFh
unconfigured LSS slave
see non-configured LSS slave
```

For more detail, please check CiA® 305 standard proposal.

friederschueler, 2024-06-14:
I will close this as answered and inactive. For such problems, a raw can log can help to identify if the problem is on the framework or device side.

closed 2024-06-14 by friederschueler