canopen-python/canopen #227: SDO block upload

effting, 2021-03-08:
I am working with a device from which I need to grab a large array of data (~7kb). For this I am using the SDO block upload feature. The client implementation looked like this:
```
instream = node.sdo[0x2002].open('rb', block_transfer=True)
outstream = io.BytesIO()
while True:
    chunk = instream.read(1024)
    if not chunk:
        break
    outstream.write(chunk)
instream.close()
outstream.truncate()
outstream.seek(0)
data = outstream.read()
```
which results in a `ValueError`:
```
Traceback (most recent call last):
  File "co_test.py", line 40, in <module>
    chunk = instream.read(1024)
  File "(...)/lib/python3.6/site-packages/canopen/sdo/client.py", line 595, in readinto
    b[:len(data)] = data
ValueError: memoryview assignment: lvalue and rvalue have different structures
```
This is caused because the data is read in segments of 7 bytes each, so after reading 146 segments (1022 bytes) the buffer has only space for 2 bytes and it tries to put in the next 7.

Quick fix for me was to set the chunk size to read in blocks of 127 * 7 = 889 bytes.

christiansandberg, 2021-03-09:
If all you want is the data read into memory you can just do:

```python
data = instream.read()
```

effting, 2021-03-19:
Yes, that would work too.

My main point is that I think the current implementation can be improved. The minimalistic approach would be to raise a better error message when the buffer size is not a multiple of 7. Actual fix would be to include an intermediate buffer in which up to 6 bytes can be stored whenever such a situation arises.

friederschueler, 2024-06-14:
Closing this due to inactivity. Feel free to contribute enhancements with a PR.

closed 2024-06-14 by friederschueler