canopen-python/canopen #229: eds LowLimit/HighLimit HEX signed/unsigned

hellmerf, 2021-03-10:
Hello,
I currently try to use an eds-file that contains serveral:
LowLimit=0x80000000
HighLimit=0x7FFFFFFF
values. When I try to set values I for example  receive  "Value 1 is less than min value 2147483648".
Looks like the hex-values are not "read" as signed ones ?
Thanks for your work ! Frank

af-silva, 2021-04-22:
Hi @hellmerf ,

I have to check, but 0x80000000 is being converted to 2147483648 and not −2147483648.
Python is converting the 0x80000000 directly, defining and unsigned interval instead of a unsigned one.

Thanks for noticing this.

friederschueler, 2023-02-17:
This issue is still occurring in the current release.  I will try to fix it with an pull request.

friederschueler, 2023-02-22:
I just added the pull request #345.

closed 2023-02-24 by af-silva

lamuguo, 2023-08-31:
May we double check the line: https://github.com/christiansandberg/canopen/blob/master/test/test_eds.py#L60

AFAIK, 0xFFFFFFFF should be -1(int32). What do you think?