canopen-python/canopen #231: TPDO for local node

mindlesstaucher, 2021-04-14:
Hi Christian,

 I try to set up a test environment where I have to simulate several canopen sensors.
So I thought I might give your canopen implementation a try. 
I manged to add 2 sensors as local nodes and my device under test is reading some sdo from it.
Now these nodes have to tramsmit a cyclic pdo. Is there a way to do that?

Thanks for helping,
Marco

By the way: great work!

af-silva, 2021-04-20:
Hi @mindlesstaucher,

Have you tried to call the `transmit()`

```python
  # the pdo object that holds the 0x60FF object, and the object it self
  self.rpdo_pointers[0x60FF].raw = self.m2counts(value)
  # transmite the pdo
  self.rpdo_pointers[0x60FF].pdo_parent.transmit()
```

gunstr, 2021-09-28:
Hi @mindlesstaucher ,

I try to do someting similar as you, but cannot even get to the point where the local node responds to the first SDO from the master asking for device type (0x1000). I guess I have missed someting fundametal in how to initialize a local slave node. Do you have any example code you can share to get me going?

Rgds
Gunnar

friederschueler, 2024-06-14:
Closing this as #47 exists.

closed 2024-06-14 by friederschueler