canopen-python/canopen #242: Need exemple for SDO server

prunith, 2021-06-14:
Hi 
I need to implement unit test for me software and for that i want to create a simulator that will respond to SDO from client.

Do you have exemple of dialogue between client and server with for exemple Vitual CAN interface?

FilipZZZ, 2022-04-01:
That's a +1 from me. I'm also looking into setting up an SDO server. It would be nice to have an example to start from.

onvav, 2022-04-01:
Hello, I would also need an example of node simulation with SDO server support. I have tried this, but the local node does not respond SDO clients requests delivered from CAN bus. It prints still the same default value loaded from EDS file, not value uploaded (local SDO server <- remote SDO client) by SDO client (canopen 2.0.0, Vector interface). Any hint will help.
`
network = canopen.Network()
network.connect(channel=0, bustype="vector", bitrate=250000)

localNode = canopen.LocalNode(3, 'test_apl_cfg_can0.eds')
localSdoServer = canopen.sdo.SdoServer(0x600 + 3, 0x580 + 3, localNode)
localSdoServer.upload(0x2222, 1)
    
try:
    while True:
        print(localNode.sdo[0x2222][1].raw)
        time.sleep(1)
except KeyboardInterrupt:
    pass
`

christiansandberg, 2022-04-01:
You shouldn't instantiate the SdoServer class directly. Always use the .sdo attribute instead.

The closest to an example for now is probably the test:
https://github.com/christiansandberg/canopen/blob/master/test/test_local.py