canopen-python/canopen #246: DS420 meaning of the Home Offset parameter?

acolomb, 2021-07-27:
When `BaseNode402.homing()` is called with `set_new_home=True` (the default!), it will try to manipulate the Home Offset (0x607C) parameter.  That's not the way the parameter is intended to work.  After a successful homing procedure, the drive should set the Actual Position (0x6063) to the Home Offset (0x607C) by itself.  By default that is zero, so the selected reference switch flank will mark the new zero position.

The library's default behavior here is backwards, and can only work with absolute position encoders.  The whole point of homing is to find a physical reference and align the logical coordinate system to it.  Trying to determine the desired offset from the value which an *unreferenced* encoder had at the physical reference point actually destroys that logical alignment.

So from my understanding, that functionality should be removed completely from the `homing()` method.  It's trivial to do from the application.  But at least it should not be the default behavior.

If this issue is viewed as legitimate, I will make a PR to fix it.

acolomb, 2021-08-02:
Opened #250 with the proposed change.

af-silva, 2021-08-16:
Valid points, PR #250 accepted. 
Closing this.

closed 2021-08-16 by af-silva