canopen-python/canopen #255: Preferred documentation style?

acolomb, 2021-08-11:
During work on the code (especially the DS402 profile), I noticed that the docstrings are often messed up.  I'm trying to find out what conventions to follow for fixing things and adding new code.  Testing locally with `python setup.py build_sphinx`.

* The `canopen.profiles.p402` module has no generated API documentation.  Is that intended?  At least the `BaseNode402` class should have docs generated IMHO, as it already contains many Sphinx-compatible docstrings.
* The `:param` and `:return` documentation sometimes uses inline types, sometimes the `:type` and `:rtype` fields.  In the one case of a `:return string: ...`, that is simply wrong syntax IIUC.  For parameters, Sphinx allows to include the type if it's only a single word.  Is that the guideline to follow?
* Are PEP-484 style type annotations an acceptable solution?  This would enable static type checkers to use the information, in addition to the Sphinx autodoc module.  If desired, I propose to use them for all return and parameter types where they are unambiguous, falling back to Sphinx directives only for ambiguous cases like "`str` or `int`".
* Sometimes there are docstrings on property *setters*, which are ignored both by the Python runtime, as well as Sphinx.  These should probably be moved to the property definition (getter)?

Sorry if these questions already have an answer somewhere, I think the package should have a small "contribution guidelines" document pointing at the relevant sources.  Having not-so-consistently documented code in the repo already adds to the confusion about what is the correct way ;-)

christiansandberg, 2021-08-12:
You're right about everything you mentioned. Fixes are most welcome! I also think we should use type annotations instead. As long as Python 3.6 is supported then it should be okay.

Almost all code were written when we had to support older versions, even 2.7, but that is no longer necessary.

acolomb, 2021-08-12:
Thanks for your reply and for quickly merging #256.

I will now start to introduce type annotations for new code whenever possible.  Maybe even go over some parts after all the DS402-related PRs are finished.

friederschueler, 2024-06-12:
Issue can be closed, PR is merged. Also we moved on and now use type annotations.

acolomb, 2024-06-12:
Thanks for the bump, I didn't even remember this one :wink:

closed 2024-06-12 by acolomb