canopen-python/canopen #269: FAULT state and Timeout when trying to change state

Cdfghglz, 2021-09-17:
Hi

Being pretty new to CANopen I am struggling to control multiple nodes. With the following code a single node works, but when connecting more, the nodes randomly go to FAULT state and canopen throws `RuntimeError('Timeout when trying to change state',)`.

_NOTE: The code below is extracted from classes so it may be a little inconsistent._

```python
network = canopen.Network()
network.connect(channel='can0', bustype='socketcan', bitrate=500000)
network.sync.start(0.15)

...

for motor in motors:
    Motor('*.dcf', node_id)

...

for motor in motors:
    motor.initialize(op_mode)
...

class Motor
    def __init__(self, dcf, network, node_id=1):
        ...
        # connect the node
        self.__node = BaseNode402(node_id, dcf)
        network.add_node(self.__node)

        self.__node.setup_402_state_machine()
        self.__node.nmt.start_node_guarding(0.01)
        ...

    def initialize(self, op_mode="PROFILED POSITION"):
        ...
        self.__node.op_mode = op_mode
        self.__command_index = self.__command_index_dict[op_mode]

        # set up the communication PDOs?
        self.__node.nmt.state = "PRE-OPERATIONAL"
        time.sleep(1)

        # TODO: is this section needed at all?
        self.__node.tpdo.read()
        self.__node.rpdo.read()
        time.sleep(0.4)

        # init the receive PDO variables
        self.__node.rpdo[self.__pdo_channel].clear()
        self.__node.rpdo[self.__pdo_channel].add_variable(0x6040)
        self.__node.rpdo[self.__pdo_channel].add_variable(self.__command_index)
        self.__node.rpdo[self.__pdo_channel].enabled = True
        self.__node.rpdo.save()
        self.__node.tpdo.save()

        time.sleep(0.4)
        self.__node.nmt.state = "OPERATIONAL"
        time.sleep(1)

        time.sleep(1)
        self.__node.state = "OPERATION ENABLED"
```

Usually the issues happen before the last line in the block above.

Could you please identify any trivial mistake I'm making? Is there any valuable resource that addresses the topic?

Thanks a lot!

af-silva, 2021-09-21:
Hi @Cdfghglz,

What is the Fault code ?? 
Is the node guard configuration on the drives well configured? 

best regards.

Cdfghglz, 2021-09-22:
Thanks @af-silva for your response.

It seems not to be quite deterministic, we've seen 8120 and 8110 motor errors.

After some debugging we seem to also be running into https://github.com/christiansandberg/canopen/issues/234. I will come back with more details as soon as we look into the 234.

af-silva, 2021-09-23:
A quick search: 
- 8110h: CAN overrun (data frame lost)
- 8120h: CAN protocol controller in error passive mode (node cannot indicate bus errors, when it is a receiver)

Check if the drivers and the CAN adapter you are using are ok...

friederschueler, 2024-06-13:
Closing this as the devices and/or can bus seems to be the problem. If you think there is a bug, feel free to open a new issue with specific information.

closed 2024-06-13 by friederschueler