canopen-python/canopen #270: TPDO data not being updated

guimpt, 2021-10-01:
Hi,
thank you very much for this library! It has been helping me a lot. I have implemented a system with 4 motors (with node id's 9, 10, 31 & 32) and as the title suggests I have some issues with the reliability of the TPDOs (using the last version of the library v1.2.1).

The problem I've seen is that for the 2 drives with higher node ids (31 & 32), the data coming from the TPDOs (triggered from a SYNC message) does not always update (is held constant for short periods of time). You can see a comparison of the same data being read from the driver with node id 10 and the driver with node id 32 in the following graph:
[image]

I have sniffed the CAN bus and data is correctly being sent after each SYNC message with updated information, so the problem must be coming from my implementation.

The configuration of the TPDOs is handled using the code you find below (for each of the nodes). Note that the reading of the actual TPDO value is handled with the function node.tpdo[register].raw and is being read asynchronously to the SYNC transmission. Between SYNC and TPDO messages some SDOs are also being sent and received from the drives.

Any help would be very much appreciated!

                self.node1.tpdo.read()
                self.node1.rpdo.read()

                # TPDO1
                self.node1.tpdo[1].clear()
                self.node1.tpdo[1].add_variable(PDO_Registers['actual_position'])
                self.node1.tpdo[1].add_variable(PDO_Registers['demand_position'])
                self.node1.tpdo[1].trans_type = 1  # Send upon sync
                self.node1.tpdo[1].enabled = True

                # TPDO2
                self.node1.tpdo[2].clear()
                self.node1.tpdo[2].add_variable(PDO_Registers['current_q_actual'])
                self.node1.tpdo[2].add_variable(PDO_Registers['current_loop_status'])
                self.node1.tpdo[2].trans_type = 1  # Send upon sync
                self.node1.tpdo[2].enabled = True

                # TPDO3
                self.node1.tpdo[3].clear()
                self.node1.tpdo[3].add_variable(PDO_Registers['sw'])
                self.node1.tpdo[3].add_variable(PDO_Registers['last_error'])
                self.node1.tpdo[3].trans_type = 1  # Send upon sync
                self.node1.tpdo[3].enabled = True

                # Unused TPDOs
                self.node1.tpdo[4].clear()
                self.node1.tpdo[4].trans_type = 1  # Send upon sync
                self.node1.tpdo[4].enabled = False

                # Save TPDOs
                self.node1.nmt.state = 'PRE-OPERATIONAL'
                self.node1.tpdo.save()

                # Change state
                self.node1.nmt.state = 'OPERATIONAL'

christiansandberg, 2021-10-01:
Do you read the PDOs from a callback? You should use [add_callback](https://canopen.readthedocs.io/en/latest/pdo.html#canopen.pdo.Map.add_callback) to get notified when a message is received.

guimpt, 2021-10-06:
Hi Christian, thanks very much for the help and quick reply!
I've changed the code so that it now uses function callbacks in order to read the variables inside the TPDO messages (following the ds402 example), but the problem does not seem to be solved. I still see the same problem on the data being read:
[image]

Do you have any other ideas as to what could be happening?

The following is the function callback as implemented and the new line added in the TPDO1 configuration:

```
    def position_callback(self, message):
        for var in message:
            if var.name == PDO_Registers['actual_position']:
                self.__actual_positionRK = var.raw
            elif var.name == PDO_Registers['demand_position']:
                self.__demanded_positionRK = var.raw
```

`
                self.node1.tpdo[1].add_callback(self.position_callback)`

guimpt, 2021-11-04:
In the end the issue was indeed related to the MCP2515, the receive buffer was not being emptied quickly enough.

Thanks to all of you for the help and ideas!