canopen-python/canopen #278: No SDO Response Received

NicoPowers, 2021-11-09:
Hello,

I have been trying to debug this error for quite some time, but it looks like it should work fine because the communication between my PC and the motor seems to be just fine, but for some reason the SDO response is not being received, even though I am receiving a heartbeat with the correct CAN ID of 1. I was able to move the motor by sending raw commands, but I would like to use the 402 profile since my motor supports it. I am using [this BLDC Harmonic Drive Motor](https://robotdigg.com/product/1757/Harmoic-reducer-BLDC-servo-motor-for-robot-arm) with [this USB2CAN device](https://www.amazon.com/Innomaker-Converter-Module-Raspberry-Zero/dp/B07P9JGXXB/ref=sr_1_1?crid=1T4DAHOG4YZH9&keywords=usb2can+innomaker&qid=1636420496&qsid=133-9019046-7959063&sprefix=usb2can+inno%2Celectronics%2C150&sr=8-1&sres=B07P9JGXXB%2CB00BWF5U0M%2CB00P0GI68M%2CB08HZ6PS61%2CB07V5JTMV9%2CB08F7NWJY5%2CB08B6F1FV5%2CB08MZZZJBM%2CB078VRJS5X%2CB00NH11N5A%2CB08JQ389FJ%2CB06XKNZT1P%2CB08PZGXMTK%2CB081VQVD3F%2CB07RV1ZCWJ%2CB06XBZ38ZJ).

My code for trying to get this to work is fairly simple:

```
    #Set CAN0 speed to 1M bps
    os.system('sudo ip link set can0 type can bitrate 1000000')
    os.system('sudo ifconfig can0 up')

    can_network = Network()
    can_network.connect(bustype='socketcan', channel='can0', bitrate = 1000000)

    node = BaseNode402(1, "YZ_MOTOR.eds")
    node.associate_network(can_network)    
    
    node.setup_402_state_machine() # ERROR: NO SOD RESPONSE RECEIVED
    
    can_network.disconnect()
    os.system('sudo ifconfig can0 down')
```

I also modified the following variables in the client.py file, of the canopen package, responsible for handling the SDO client communication (just to see if that would fix it):

```
    #: Max time in seconds to wait for response from server
    RESPONSE_TIMEOUT = 0.5

    #: Max number of request retries before raising error
    MAX_RETRIES = 10

    #: Seconds to wait before sending a request, for rate limiting
    PAUSE_BEFORE_SEND = 0.1
```

This is what my console output looks like. As you can see, I do received a heartbeat and it reads from my .eds file just fine. But for some reason it cannot receive any SDO Response, and I do remember receiving a response from 0x581 when I manually sent raw commands to modify the target position of the motor by subscribing to 0x581 (using the basic Network class) and performing some logic.

[image]

Any help is much appreciated, thanks!

acolomb, 2021-11-09:
The obvious thing is that your code uses `node.associate_network()` instead of `network.add_node(node)`.  But I think that should still work, although the examples always do the latter.

What version are you using? Can you get a trace of messaged using e.g. `candump`? Preferably with timestamps on.

NicoPowers, 2021-12-13:
Sorry for the late reply, i have been busy with senior design.

This is output of the candump:
[image]

So I am getting the heartbeat and its trying to talk to it and it looks like I am getting a response, so I don't know why it's saying no SDO response?

acolomb, 2022-01-15:
The node answers to the first three SDO requests (1400:01 to 1400:03), but stops responding when asked about 1400:05.  Check your EDS and device manual whether that it actually supported.  Some devices, instead of returning an SDO abort error code, just fail to answer when not understanding a request.

friederschueler, 2024-06-14:
Closing this, as there is no follow up response and the problem is very old.

closed 2024-06-14 by friederschueler