canopen-python/canopen #283: Code 0x06040041, Object cannot be mapped to the PDO

WangRong11, 2021-11-12:
Hello!
When I debug the code, I encounter such a problem. Can you help me analyze the reason?
This is my code:
```
from re import S
import canopen
import time

network = canopen.Network()

# node = canopen.RemoteNode(1, '/home/tx2/communication_ws/src/canopen/examples/eds/hcfaCAN_v1_4_R1.eds')
node = canopen.RemoteNode(1, '/home/tx2/communication_ws/src/canopen/examples/eds/401.eds')
network.add_node(node)

network.connect(bustype='socketcan', channel='can0')

node.tpdo.read()
node.rpdo.read()

# # 重映射 TPDO[1]
node.tpdo[1].clear()
# node.tpdo[1].add_variable('Statusword')#6041
# node.tpdo[1].add_variable('Velocity actual value')#606C 
node.tpdo[1].add_variable('Read input 8 bit','digital Inputs 1-8')
node.tpdo[1].trans_type = 254
node.tpdo[1].event_timer = 10
node.tpdo[1].enabled = True

# node.rpdo[1].clear()
# node.rpdo[1].add_variable('Write output 8-bit','Digital Outputs 1-8')
# node.rpdo[1].enabled = True

node.nmt.state = 'PRE-OPERATIONAL'
node.tpdo[1].cob_id = node.tpdo[1].predefined_cob_id
# node.rpdo[1].cob_id = node.rpdo[1].predefined_cob_id

node.tpdo[1].save()
# node.rpdo[1].save()

# node.rpdo[1]['Write output 8-bit.Digital Outputs 1-8'].raw = 0x00
# node.rpdo[1]['Target Position'].raw = 1000
# node.rpdo[1].start(0.1)

node.nmt.state = 'OPERATIONAL'

node.tpdo[1].wait_for_reception()
# speed = node.tpdo[1]['Velocity actual value'].raw
speed = node.tpdo[1]['Read input 8 bit.digital Inputs 1-8'].raw

network.disconnect()
```
When the code node.tpdo[1].save() is run, an error will be reported

acolomb, 2021-11-12:
Can you try to enable logging and tell where it goes wrong exactly? The `.save()` call does a whole bunch of SDO transfers, so not clear what exactly fails.

From the error code it should be pretty clear, so you need to check your datasheet or possibly the EDS file whether this digital inputs register can actually be mapped to PDO. Maybe you got the wrong one, are there other subindex entries concerning digital inputs?

WangRong11, 2021-11-12:
Logging:
```
DEBUG:canopen.pdo:TPDO Map as 2
DEBUG:canopen.pdo:RPDO Map as 2
DEBUG:can:can config: {'channel': 'can0', 'interface': 'socketcan'}
INFO:can.interfaces.socketcan.socketcan:Created a socket
DEBUG:can.interfaces.socketcan.socketcan:Binding socket to channel=can0
DEBUG:can.interfaces.socketcan.socketcan:Bound socket.
INFO:canopen.network:Connected to 'socketcan channel 'can0''
DEBUG:canopen.sdo.client:Reading 0x1800:1 from node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S                DLC:  8    40 00 18 01 00 00 00 00
DEBUG:canopen.variable:Value of Transmit PDO Communication Parameter 1.COB-ID (0x1800:1) is 4026532225
cob_id 4026532225
INFO:canopen.pdo.base:COB-ID is 0x10000181
INFO:canopen.pdo.base:PDO is disabled
INFO:canopen.pdo.base:RTR is not allowed
DEBUG:canopen.sdo.client:Reading 0x1800:2 from node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S                DLC:  8    40 00 18 02 00 00 00 00
DEBUG:canopen.variable:Value of Transmit PDO Communication Parameter 1.Transmission Type (0x1800:2) is 254
INFO:canopen.pdo.base:Transmission type is 254
DEBUG:canopen.sdo.client:Reading 0x1800:3 from node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S                DLC:  8    40 00 18 03 00 00 00 00
DEBUG:canopen.variable:Value of Transmit PDO Communication Parameter 1.Inhibit Time (0x1800:3) is 1000
INFO:canopen.pdo.base:Inhibit time is set to 1000 ms
DEBUG:canopen.sdo.client:Reading 0x1800:5 from node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S                DLC:  8    40 00 18 05 00 00 00 00
DEBUG:canopen.variable:Value of Transmit PDO Communication Parameter 1.Event Timer (0x1800:5) is 0
INFO:canopen.pdo.base:Event timer is set to 0 ms
INFO:canopen.pdo.base:Could not read SYNC start value ('Subindex 6 was not found')
DEBUG:canopen.sdo.client:Reading 0x1A00:0 from node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S                DLC:  8    40 00 1a 00 00 00 00 00
DEBUG:canopen.variable:Value of TPDO Mapping Parameter 1.Number of Entries (0x1A00:0) is 0
DEBUG:canopen.sdo.client:Reading 0x1801:1 from node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S                DLC:  8    40 01 18 01 00 00 00 00
DEBUG:canopen.variable:Value of Transmit PDO Communication Parameter 2.COB-ID (0x1801:1) is 4026532481
cob_id 4026532481
INFO:canopen.pdo.base:COB-ID is 0x10000281
INFO:canopen.pdo.base:PDO is disabled
INFO:canopen.pdo.base:RTR is not allowed
DEBUG:canopen.sdo.client:Reading 0x1801:2 from node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S                DLC:  8    40 01 18 02 00 00 00 00
DEBUG:canopen.variable:Value of Transmit PDO Communication Parameter 2.Transmission Type (0x1801:2) is 254
INFO:canopen.pdo.base:Transmission type is 254
DEBUG:canopen.sdo.client:Reading 0x1801:3 from node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S                DLC:  8    40 01 18 03 00 00 00 00
DEBUG:canopen.variable:Value of Transmit PDO Communication Parameter 2.Inhibit Time (0x1801:3) is 1000
INFO:canopen.pdo.base:Inhibit time is set to 1000 ms
DEBUG:canopen.sdo.client:Reading 0x1801:5 from node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S                DLC:  8    40 01 18 05 00 00 00 00
DEBUG:canopen.variable:Value of Transmit PDO Communication Parameter 2.Event Timer (0x1801:5) is 0
INFO:canopen.pdo.base:Event timer is set to 0 ms
INFO:canopen.pdo.base:Could not read SYNC start value ('Subindex 6 was not found')
DEBUG:canopen.sdo.client:Reading 0x1A01:0 from node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S                DLC:  8    40 01 1a 00 00 00 00 00
DEBUG:canopen.variable:Value of TPDO Mapping Parameter 2.Number of Entries (0x1A01:0) is 0
DEBUG:canopen.sdo.client:Reading 0x1400:1 from node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S                DLC:  8    40 00 14 01 00 00 00 00
DEBUG:canopen.variable:Value of Receive PDO Communication Parameter 1.COB-ID (0x1400:1) is 513
cob_id 513
[... 58 lines cut ...]
[... cut ...]
DEBUG:canopen.nmt:Received heartbeat can-id 1793, state is 127
DEBUG:canopen.nmt:Received heartbeat can-id 1793, state is 127
DEBUG:canopen.nmt:Received heartbeat can-id 1793, state is 127
DEBUG:canopen.nmt:Received heartbeat can-id 1793, state is 127
DEBUG:canopen.nmt:Received heartbeat can-id 1793, state is 127
DEBUG:canopen.nmt:Received heartbeat can-id 1793, state is 127
DEBUG:canopen.nmt:Received heartbeat can-id 1793, state is 127
DEBUG:canopen.nmt:Received heartbeat can-id 1793, state is 127
```

acolomb, 2021-11-13:
Thanks for the log.  It doesn't actually contain the error message though.  Apparently there is some successful SDO communication before the failure, so the message is quite clearly caused by the device just not supporting this object mapped to a PDO.

Usually the device manual or the EDS should indicate which objects can be PDO-mapped.  Double-check whether you have the right object index for the purpose.

WangRong11, 2021-11-16:
Thank you for your reply. After listening to your suggestions, I solved this problem.

closed 2021-11-16 by WangRong11