canopen-python/canopen #286: Local node as heartbeat consumer

gunstr, 2021-11-29:
Hi,
I have implemented a local slave node with this excellent library and it's running well. However, I need the node to reset some outputs in case of loosing the connection to the master (such as a cut on the cable). For this I was thinking to set the local node as a heart beat consumer, but I cannot figure out how to do. What I understand a local node is always associated with an NMTSlave and it seems it can only be set up as a heartbeat producer. Is this correct and if so, do you have any suggestion for an alternative solution to act on lost communication?
Best regards

christiansandberg, 2021-11-29:
How do you send commands to the local node? If the node expects PDO messages then you could detect when those messages stops arriving. You could probably do the same for other services.

The local node support is far from complete. I don't think heartbeats are designed to be sent from the master however, but I could be wrong.

gunstr, 2021-11-29:
An RPDO is used to send commands to the node, but to limit bus load it's only sent on transitions. As the detection of lost communication is important only when the node is activated (=moving) I will see if I can send it in continues mode for activated devices.

I'm fully aware that the local node support is not complete and given that I'm surprised how well it works. I'm sure I will find a workaround, but just wanted to check with you experts as the documentation of local node use case is limited.

What I understand it is recommended that all nodes (masters as well as slaves) should support both heartbeat producing and consuming. Only exception is simple sensors not supporting RPDO at all.  I have not found the actual recommendation, but there is an [article in the subject](https://can-newsletter.org/uploads/media/raw/dfb1993721f0a1cc96533c47adaadd7e.pdf) published by CiA.

I use a Codesys based master that can be configured as a heartbeat producer, so on that end it works.

af-silva, 2021-12-03:
Ok, my understanding and what I used was configuring the NMT to periodically send a heartbeat through the `node.nmt.start_node_guarding(0.01)` from Master to the remote nodes. This would enable us to monitor the well being of the master (controller) node and if it fails all the remote nodes would stop (this must be configured on the remote node). Additionally, if one of the nodes stop replying (e.g., you can picture for instance four motors of a car) the master also would know and you could handle what to do in this case (e.g., halt all the motors). 

[image]
Image credits: infosys.beckhoff.com

In my use case the remote nodes were motors, and I implemented the master and configured the heartbeat on both master and slave nodes. Look for `start_node_guarding`, `on_heartbeat` and `add_hearbeat_callback`. I never used the LocalNode, so I have limited knowledge on how it handles/maintains the state of the heartbeat but the mechanism is explained in the image above, and is very useful for this kinds of situation.

Hope this can help you in some way.

gunstr, 2021-12-04:
Thanks for your comment. I actually use something in line with what you have done in the two masters I have implemented in the same system. The challenge with the LocalNode is that the [NMTslave](https://github.com/christiansandberg/canopen/blob/a83a6b83b1589f493b2205042ae114053d26f61e/canopen/nmt.py#L192) currently only seem to support the role as heartbeat producer, there are no `on-heartbeat` or `add_heartbeat_callback` methods.

I have solved my problem by adding a "toggle-bit" to the rpdo that the master toggles each 100ms when it has set a slave to active mode, This will force the master to send the rpdo with this interval. I have then implemented a timeout in the LocalNode slave if an rpdo is not received as expected when it is active. Maybe not the most efficient solution, but it works as intended for now.

I do not feel I have enough knowledge in Python or CanOpen yet, but if I get a bit more time in the future I might clone the project and see if I can figure out how to add a bit more functionality to the LocalNode -  would be interesting.

pierreSCpapon, 2022-02-18:
I might be running in the same issue as @gunstr 
I am working with a LINAK A36 [CANopen datasheet](https://cdn.linak.com/-/media/files/ic-and-bus-actuators/techline-canopen-user-manual-eng.pdf)
From my understanding, it requires a heartbeat from master every 100ms
[image]
CANID 701 with payload of 0x05 (I understand this as master node 01 in operational mode)

I think this is what node guarding should do as mentioned by @af-silva but the payload seems to be empty and if I understood correctly from the datasheet, I need 0x05 as payload.

Any ideas what I can do about this?

af-silva, 2022-02-18:
Hi @pierreSCpapon ,

> **Heartbeat protocol**
>
>The Heartbeat protocol is a cyclically transmitted message that informs all heartbeat consumers of the availability of the heartbeat producer. In addition to the availability of the heartbeat producer, the heartbeat protocol provides the current NMT FSA state of the heartbeat producer. The cycle time for transmitting the Heartbeat message is configurable in the object dictionary index 1017h.
>
>**Node-/Life guarding protocol**
>
> 
> Guarding is an outdated method of checking whether a guarded CANopen device is still working in the correct network state. As this is an RTR-based service, the Heartbeat protocol is used in new designs. In old-fashioned applications, the host controller triggers the error control information of the monitored CANopen devices via the CAN remote frame (RTR) for example. Each monitored CANopen device has to be triggered individually. The monitored device replies with a CAN data frame that indicates the current NMT FSA state. `
>
[Source](https://www.can-cia.org/can-knowledge/canopen/error-control-protocols/)

There are two ways to do this, and it depends how you have things configured, who is the master (software or hardware) and who are the remote nodes (software or hardware). For what I could understand you have created a master and want to communicate with remote nodes, right?
Right now I think we only have the Node Guarding protocol implemented to check on the health of the remote nodes, it sends a periodic message asking the slave node to send its NMT Status.
We can implement an Heartbeat producer at the `NmtMaster(NmtBase)` similar to the `start_node_guarding` but instead of not sending any data we send the NMT Status of the master node (Pre-operational state (0x7F), Operational state (0x05), or Stopped state (0x04))

You can make the changes and do a PR,
If I have some time a can implement it during the weekend.