canopen-python/canopen #294: Support for "USB to CAN Bus Converter Adapter serial port TO CAN /RS232 TO CAN With TVS"?

yanone, 2022-01-26:
Forgive my issue here. I'm completely new to CAN bus and don't know where to start.

Will this USB device work for me on a Raspberry Pi with `canopen`?:
https://www.ebay.de/itm/265203036092?hash=item3dbf552fbc:g:7R8AAOSw3wFg0Ea4

Thank you

gedass2000, 2022-01-26:
Hi ,
I use TouCAN USB<->CAN adapter on a Raspberry Pi  and SocketCAN interface.

acolomb, 2022-01-26:
I suppose it will be difficult. They seem to provide only Windows drivers, and there is no standard for CAN interfaces on that OS. They also talk about a virtual COM port, so it will show up as a serial port device on the computer. If you're lucky, they "borrowed" the API / protocol from some other manufacturer's CAN adapter, and you could use some existing driver from the python-can project (which provides the unelderlying communication layer for this library). But it may as well some custom protocol that nobody knows, then you're on your own.

yanone, 2022-01-26:
> I use TouCAN USB<->CAN adapter on a Raspberry Pi and SocketCAN interface.

That's a good lead, thank you. How is the connectivity in terms of cabling? I'm not sure yet which other devices on my bus I'll talk to, as I'm just now researching the basics, but I'm expecting to need just CANH and CANL pins. Are these available as-is on the RS232 socket?

acolomb, 2022-01-26:
Yes, they are.  See for example https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z0000019YIGSA2&l=en-US

You may want to connect CAN-GND as well, if the devices don't have a common power supply. But H and L is usually enough for a start.

The TouCAN is a fine device, I bought several as well and it works great with Linux and after some DLL hassles also on Windows.

yanone, 2022-01-27:
Thank you all. That's good information, and I'll close this issue now.

closed 2022-01-27 by yanone