canopen-python/canopen #295: Mapping the sub-indiced objects

Siverian, 2022-01-28:
Hello,
I am looking to work with a absolute resolver from Wachendorff, controlled with CANopen. I have been able to put the resolver into 'OPERATIONAL' mode and read Vendor and Device name back. So, in principal, the communication works and .eds file is fine, at least some part.

Reading with SDO makes no problem. Values are fine and no error appears. 
The problem makes reading, using  PDO. The Position object is not divided by sections and **add_variable('Position Value', length=32')** works fine. But object Speed value is divided by 2 sub's. If it try to add this object in this way:

```
node.tpdo[1].clear()
node.tpdo[1].add_variable('Speed value')
node.tpdo[1].trans_type = 254
node.tpdo[1].enabled = True
node.tpdo.save()
```
I get the error:
`canopen.sdo.exceptions.SdoAbortedError: Code 0x06040041, Object cannot be mapped to the PDO`

It helps to add variable using this syntax:
<span style="color:red"> `node.tpdo[1].add_variable('Speed value','Speed value')` </span>.

But when I try to read the result in this way:
```
# Re-map TxPDO1
node.tpdo[1].clear()
node.tpdo[1].add_variable('Speed value','Speed value')
node.tpdo[1].trans_type = 254
node.tpdo[1].enabled = True
node.tpdo.save()

node.nmt.state = 'OPERATIONAL'
time.sleep(0.05)
speed = node.tpdo['Speed value'].phys
```
I get this error:
`KeyError: 'PDO: Speed value was not found in any map'`

Declaration in this way solves the problem:
`speed = node.tpdo[0x6030].phys`

The question is how to map this variable in the right way?
The code is here: 
```
import canopen
import time

# import logging
# logging.basicConfig(level=logging.DEBUG)

network = canopen.Network();
network.connect(bustype='ixxat', channel=0, bitrate=250000)

node = network.add_node(127, 'wdga-st-co_2013-04-22.eds')

node.nmt.state = 'PRE-OPERATIONAL'
device_name = node.sdo['Manufacturer device name'].raw
# vendor = node.sdo[0x1018][1].raw
vendor = node.sdo['Identity Object'][1].raw
Pos = node.sdo['Position Value'].raw
SP = node.sdo['Speed value'][1]
ACC = node.sdo['Acceleration value'][1]

print("Device name: %s" % device_name)
print("Vendor ID: %d" % vendor)
print("Speed %d" % SP.phys)
print("Position %d" % Pos)

speed_obj = node.object_dictionary['Speed value'][2]
node.tpdo.read()

# Re-map TxPDO1
node.tpdo[1].clear()
node.tpdo[1].add_variable('Speed value','Speed value')
node.tpdo[1].trans_type = 254
node.tpdo[1].enabled = True
node.tpdo.save()

node.nmt.state = 'OPERATIONAL'
time.sleep(0.05)
speed = node.tpdo['Speed value'].phys

print("Speed2 ", speed)

network.disconnect()
```
The object file:
```
;--- 6004h: Position value ------------------------------------------

[6004]
ParameterName=Position Value
ObjectType=0x7
DataType=0x0007
AccessType=ro
PDOMapping=1
ObjFlags=0x0
;--- 6030h: Speed value ---------------------------------------------

[6030]
ParameterName=Speed value
ObjectType=0x8
SubNumber=2
ObjFlags=0x0

[6030sub0]
ParameterName=Highest sub-index supported
ObjectType=0x7
DataType=0x0005
AccessType=ro
DefaultValue=0x01
PDOMapping=0
LowLimit=1
HighLimit=1
ObjFlags=0

[6030sub1]
ParameterName=Speed value
ObjectType=0x7
DataType=0x0003
AccessType=ro
PDOMapping=1
ObjFlags=0
```

acolomb, 2022-01-28:
I think the mapping part is fine. But you probably need to specify the subindex again when getting the value from the PDO. IIRC it needs to be appended with a " - " delimiter. Not quite sure as I'm currently traveling and answering off the top of my head.

You can iterate through the PDO to see what fields are contained and what name is assigned.

christiansandberg, 2022-01-28:
You should access subindices using . notation:

```
val = node.tpdo['Some group.Some subindex'].raw
```

acolomb, 2022-01-31:
Did you try iterating over the object to see which keys are contained?

Siverian, 2022-02-03:
I'm not sure how to iterate correctly. After import I see Object "Speed value" with 2 elements 
But indexing using 'Speed value.Speed value' does not help

[image]

acolomb, 2022-02-03:
Did you look at the subindices attribute?

Sorry I don't know what you did to get that pictured tree view.  Which object is that?

I was thinking along the lines of

    for v in node.tpdo[1]:
        print(v)

Are you maybe missing the `[1]` to access the **first** TPDO?  The `.tpdo` member is a map of all configured TPDOs.

friederschueler, 2024-06-14:
I am closing this now, as there is no further response and it should be fixed with

    val = node.tpdo[1]['Speed value.Speed value'].raw
    
as acolomb described.

closed 2024-06-14 by friederschueler