canopen-python/canopen #297: Python 2.7 compatibility

mzahana, 2022-03-07:
Hi

Thanks for the great package!

I need to use this package with Python 2.7 (because I am using ROS melodic with Ubuntu 18, with no much of choice to use Python 3 without going through extensive dependency issues). Is it possible to use the latest version with Python 2.7?

Thanks

mzahana, 2022-03-07:
@christiansandberg  thanks a lot!

Is the 402 profile implemented in v1.1.0?
Any critical changes between the latest version and v1.1.0?

christiansandberg, 2022-03-07:
The 402 profile was implemented in 1.1.0 but it has gone through quite some changes since then. I don't know the specifics so you need to look at the release notes here in Github.

mzahana, 2022-03-08:
@christiansandberg  I downgraded to v1.1.0 and I get the following error.
```
node.state = 'SWITCH ON DISABLED'
  File "/root/.local/lib/python2.7/site-packages/canopen-1.1.0-py2.7.egg/canopen/profiles/p402.py", line 442, in state
    if _change_state(next_state):
NameError: global name '_change_state' is not defined
```

Any idea what is wrong here?

christiansandberg, 2022-03-08:
I wish I could help out more but since you need to use unsupported versions you're basically on your own. You may have to fork the source code from v1.1.0 and do any bugfixes yourself. 🙁

mzahana, 2022-03-08:
@christiansandberg Thanks!
Problem solved. I forked the repo, tweaked the v1.2.0 a bit, and it worked fine with Python 2.7.

af-silva, 2022-03-08:
Hi @mzahana 

>Is the 402 profile implemented in v1.1.0?

The DS402 profile should work fine in the v1.1.0 with ROS and python 2.7. I had it working with ROS Melodic. The ROS implementation however is application specific and so not present here.

>Any critical changes between the latest version and v1.1.0?

The "first" versions of the DS402 profile make some assumptions about how the states transitions were handled, since then some trim was done to the DS402 profile to not specify any assumptions, and let the application handled those transitions and how it should behave. 
The example provided is a good starting point, if you create a simple ROS node and extend the BaseNode402 to a specific implementation that handles the needed topics to receive and transmit information you should be able to get things working very fast. 

Good luck :) any doubts, leave a comment ;)
PS: Can I close this thread?

mzahana, 2022-03-09:
@af-silva Thank you for your response. 
My issue in `v1.1.0` was that sending target speed for a particular node using `rpdo[1]` was not consumed. However, when I switched to `v1.2.0` , it worked. Here is the code snippet I use
```python
try:
   node = self._network[node_id]
   node.rpdo[1]['Target speed'].phys = vel
except Exception as e:
   logging.error("Could not set velocity for node {}. Error: ".format(node_id, e))`
```

The same code didn't work with `v1.1.0`, but worked with `v1.2.0`. However, I am not sure what the issues is in `v1.1.0`

Feel free to close the issue.

closed 2022-03-09 by af-silva