canopen-python/canopen #300: Connection Problems from Raspberry Pi to Ingenia Board via Canopen

CptNugai, 2022-03-09:
Hello,
First i want to say that I'm completely new on working with CANOpen.
I'm currently trying to establish a connection from my Raspberry Pi over the 2-CH-CAN HAT (https://www.waveshare.com/wiki/2-CH_CAN_HAT) to an Ingenia Neptune board (https://ingeniamc.com/support/neptune/). The communication should be done with the canopen package for python 3. I tried to implement a basic code by following the examples given on https://canopen.readthedocs.io/en/latest/network.html . Further I used the provided .eds file by Ingenia.

` import canopen
import time

network = canopen.Network()

RPi = canopen.LocalNode(0x19, '/home/pi/python/RPi.eds') #master node
network.add_node(RPi)

Motor3 = canopen.RemoteNode(0x22, '/home/pi/python/Motor3.eds') #slave node
network.add_node(Motor3)
network.connect(bustype='socketcan', channel='can0', bitrate=100000, can_filters=None)
Motor3.nmt.state = 'OPERATIONAL'

network.scanner.search()
time.sleep(0.05)
for node_id in network.scanner.nodes:
    print("Found node %d!" % node_id)

for node_id in network:
    print('test2')
    print(network[node_id])

vendor_id = Motor3.sdo[0x1018][1].raw
print(vendor_id)

network.disconnect() `

After executing this code I get the follwoing result and errors.
` test2
<canopen.node.local.LocalNode object at 0x75b75c58>
test2
<canopen.node.remote.RemoteNode object at 0x759afdf0>
Transfer aborted by client with code 0x05040000
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/canopen/sdo/client.py", line 66, in read_response
    response = self.responses.get(
  File "/usr/lib/python3.9/queue.py", line 179, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/python/main.py", line 27, in <module>
    vendor_id = Motor3.sdo[0x1018][1].raw
  File "/usr/local/lib/python3.9/dist-packages/canopen/variable.py", line 75, in raw
    value = self.od.decode_raw(self.data)
  File "/usr/local/lib/python3.9/dist-packages/canopen/variable.py", line 37, in data
    return self.get_data()
  File "/usr/local/lib/python3.9/dist-packages/canopen/sdo/base.py", line 128, in get_data
    return self.sdo_node.upload(self.od.index, self.od.subindex)
  File "/usr/local/lib/python3.9/dist-packages/canopen/sdo/client.py", line 117, in upload
    fp = self.open(index, subindex, buffering=0)
  File "/usr/local/lib/python3.9/dist-packages/canopen/sdo/client.py", line 205, in open
    raw_stream = ReadableStream(self, index, subindex)
  File "/usr/local/lib/python3.9/dist-packages/canopen/sdo/client.py", line 251, in __init__
    response = sdo_client.request_response(request)
  File "/usr/local/lib/python3.9/dist-packages/canopen/sdo/client.py", line 85, in request_response
    return self.read_response()
  File "/usr/local/lib/python3.9/dist-packages/canopen/sdo/client.py", line 69, in read_response
    raise SdoCommunicationError("No SDO response received")
canopen.sdo.exceptions.SdoCommunicationError: No SDO response received `

It seems like that there is no connection created to the connected node.
This output raises some questions:
- Is it necessary to define a master node? If yes, is it the Raspberry Pi and which .eds file is needed? Currently I use the same .eds file as used for the Ingenia board (the node ID is changed), because I couldn't find a specific one for the Raspberry Pi or the 2-CH-CAN-HAT.
- How far has the connection to be established that the command 'network.scanner.search()' finds any nodes?

It would be nice if someone could help me solving my problems.

Best regards

acolomb, 2022-03-09:
Looks like you are having general CAN communication problems.  Have you verified polarity and termination resistors?  Does the baud rate match on both sides?  Is the HAT initialized correctly?  I've had some experience with the other Waveshare CAN HAT, where the used 12 MHz crystal would not allow 1000 kbit baud rate for example.

> * Is it necessary to define a master node? If yes, is it the Raspberry Pi and which .eds file is needed? Currently I use the same .eds file as used for the Ingenia board (the node ID is changed), because I couldn't find a specific one for the Raspberry Pi or the 2-CH-CAN-HAT.

No, you don't need that `LocalNode` stuff at all for your purpose AFAICT.

> * How far has the connection to be established that the command 'network.scanner.search()' finds any nodes?

Physical layer needs to be working, and the devices need to support the protocol (LSS probably).  I suggest you first start out trying to get basic CAN communication working using `candump` and `cansend` commands on both ends.  Only if that is verified to work, go up one layer and start using CANopen.

acolomb, 2022-03-22:
Sorry for not responding earlier.  So you did get the problem fixed?

For the record, yes bitrate and baudrate have the same meaning in this context.