canopen-python/canopen #301: Import/export of BaudRate_xx in EDS not working

Sudlav, 2022-03-14:
According to CiA 306 the valid baudrate keywords in the DeviceInfo section are:
BaudRate_10, BaudRate_20, BaudRate_50, etc

However, these are not recognized by the import_eds() method, instead import_eds() recognizes:
Baudrate_10, Baudrate_20, Baudrate_50, etc
(note the casing of "R"/"r")
Also the export_eds() record will export the names as:
Baudrate_10, Baudrate_20, Baudrate_50, etc

This means that if I import an EDS file and then just export it to a new filename this information gets lost.

af-silva, 2022-03-14:
Hi @Sudlav,

After a quick search it seems that yes the DeviceInfo should have BaudRate_XX, i.e., in camel case.
You can make a PR with the changes needed to support this, or wait and maybe during the weekend I can have a look. 

Thanks

af-silva, 2022-03-18:
fix by #303  and #304

closed 2022-03-18 by af-silva