canopen-python/canopen #309: dsp 402 Exception "unhandled AssertionError" OPERATIONAL

superlogan72, 2022-04-08:
I'm trying to make communication with servo drive,
I write this few line:
node = BaseNode402(1, 'EBS.eds')
network = canopen.Network()
network.add_node(node)
network.connect(bustype="gs_usb", channel=dev.product, bus=dev.bus, address=dev.address, bitrate=250000)
node.setup_402_state_machine()
after this line i get the error Exception "unhandled AssertionError" OPERATIONAL
Can you help me

superlogan72, 2022-04-11:
nodo_luca = BaseNode402(1, 'EBS.eds')
network = canopen.Network()
network.add_node(nodo_luca)
network.connect(bustype="gs_usb", channel=dev.product, bus=dev.bus, address=dev.address, bitrate=500000)
network.scanner.search()
time.sleep(0.05)
for node_id in network.scanner.nodes:
    print("Found node %d!" % node_id)
    #here i get Found node 1!
node.setup_402_state_machine()
#i get:
Exception "unhandled AssertionError"
OPERATIONAL
File: C:\Users\Luca\AppData\Local\Programs\Python\Python38\lib\site-packages\canopen\profiles\p402.py, Line: 239

acolomb, 2022-04-11:
Can you check what the `.nmt.state` attribute shows after the scanner has found the node? You may need to explicitly switch it to PRE-OPERATIONAL or at least wait until it is no longer initializing before trying the setup routine.

acolomb, 2022-04-11:
That looks a bit better, but the 0x3000 range numbers don't make any sense. Those are in the manufacturer specific range and not used by the DS402 code at all.

superlogan72, 2022-04-12:
hello Christian
can you help me to make it in the right way, I have the eds file of the
device, I'm also ready to pay your work
Regards
Luca

superlogan72, 2022-04-12:
yes is better but in any case i'm not able to move the motor

acolomb, 2022-04-12:
Well you need to find out where these error messages are coming from.  We can't help you with that with so little information.  As I said, the DS402 code in this library does not touch any `0x3***` object.

superlogan72, 2022-04-13:
i see the the message came from the class ObjectDictionary in the __getitem__ function but i don't understand why
class ObjectDictionary(MutableMapping):
    """Representation of the object dictionary as a Python dictionary."""

    def __init__(self):
        self.indices = {}
        self.names = {}
        self.comments = ""
        #: Default bitrate if specified by file
        self.bitrate: Optional[int] = None
        #: Node ID if specified by file
        self.node_id: Optional[int] = None
        #: Some information about the device
        self.device_information = DeviceInformation()

    def __getitem__(
        self, index: Union[int, str]
    ) -> Union["Array", "Record", "Variable"]:
        """Get object from object dictionary by name or index."""
        item = self.names.get(index) or self.indices.get(index)
        if item is None:
            name = "0x%X" % index if isinstance(index, int) else index
            raise KeyError("%s was not found in Object Dictionary" % name)
        return item

superlogan72, 2022-04-14:
Hello again now we reach the driver producers he reply asking us to test the following command
 In velocity
     RxPDO
    sPdo.dwMap[0] = 0x60400010; // Control register
    sPdo.dwMap[1] = 0x60FF0020; // Velocity set-point

     TxPDO :          
     sPdo.dwMap[0] = 0x60410010; // Status register
     sPdo.dwMap[1] = 0x60640020; // Position

    if you work with Sync. PDO you need to send the SYNC (0x80 any 2 milliseconds

How can I send this command with pythoncan

acolomb, 2022-04-15:
> i see the the message came from the class ObjectDictionary in the **getitem** function but i don't understand why

That's the place where the error is raised.  But what you need to find out (if these errors are actually a problem) is where the call comes from, e.g. the full backtrace (function call stack) when the message is printed.

For mapping objects to the PDO, see the documentation at https://canopen.readthedocs.io/en/latest/pdo.html. The following chapter describes how to use the SYNC service as well.

superlogan72, 2022-04-26:
Hello i have proceed with some test and basing to the example I have done this:
import canopen
import sys
import os
import traceback
import usb
import time

try:

    # Start with creating a network representing one CAN bus
    network = canopen.Network()
    dev = usb.core.find(idVendor=0x1D50, idProduct=0x606F)

    # Connect to the CAN bus
    network.connect(bustype="gs_usb", channel=dev.product, bus=dev.bus, address=dev.address, bitrate=500000)

    network.check()

    # Add some nodes with corresponding Object Dictionaries
    node = canopen.BaseNode402(1, 'TBS3.eds')
    network.add_node(node)
    # network.add_node(34, 'eds/example34.eds')
    # node = network[34]

    # Reset network
    node.nmt.state = 'RESET COMMUNICATION'
    #node.nmt.state = 'RESET'
    node.nmt.wait_for_bootup(15)

    print('node state 1) = {0}'.format(node.nmt.state))
    # Iterate over arrays or records
    error_log = node.sdo[0x1003]
    for error in error_log.values():
        print("Error {0} was found in the log".format(error.raw))

    for node_id in network:
        print(network[node_id])

    print('node state 2) = {0}'.format(node.nmt.state))

    # Transmit SYNC every 100 ms
    network.sync.start(0.002)

    node.load_configuration()

    print('node state 3) = {0}'.format(node.nmt.state))

    node.setup_402_state_machine()

    device_name = node.sdo[0x1008].raw
    vendor_id = node.sdo[0x1018][1].raw

    print(device_name)
    print(vendor_id)
    
    node.state = 'SWITCH ON DISABLED'

    print('node state 4) = {0}'.format(node.nmt.state))
    
    node.tpdo[1].clear()
    node.tpdo[1].add_variable(0x6041)
    node.tpdo[1].add_variable(0x6061)
    node.tpdo[1].add_variable(0x606C)
    node.tpdo[1].trans_type = 1
    node.tpdo[1].enabled = True
    node.tpdo.save()
    node.rpdo[1].clear()
    node.rpdo[1].add_variable(0x6040)
    node.rpdo[1].add_variable(0x6060)
    node.rpdo[1].add_variable(0x60FF)
    node.rpdo[1].enabled = True
    node.rpdo.save()
    time.sleep(1)
    
    node.rpdo.read()
    time.sleep(1)
    node.rpdo[1][0x6060].raw = 3 #mode velocity
    node.rpdo[1].transmit()
    time.sleep(1)
    node.rpdo[1][0x6040].raw = 0x06  #enable operation and quick stop (active lower)
    node.rpdo[1].transmit()
    time.sleep(1)
    node.rpdo[1][0x6040].raw = 0x07 #enable operation and quick stop (active lower), Switch on
    node.rpdo[1].transmit()
    time.sleep(1)
    node.rpdo[1][0x6040].raw = 0x0F  #enable operation and quick stop (active lower), Switch on,Disable Voltage
    node.rpdo[1].transmit()
    time.sleep(1)
    node.rpdo[1][0x60FF].raw = 0x200 #Velocity set point
    node.rpdo[1].transmit()
    time.sleep(1)
    node.state = 'READY TO SWITCH ON'
    node.state = 'SWITCHED ON'

    print('Node booted up')

    timeout = time.time() + 15
    node.state = 'READY TO SWITCH ON'
    while node.state != 'READY TO SWITCH ON':
        if time.time() > timeout:
            raise Exception('Timeout when trying to change state')
        time.sleep(0.001)

    timeout = time.time() + 15
    node.state = 'SWITCHED ON'
    while node.state != 'SWITCHED ON':
        if time.time() > timeout:
            raise Exception('Timeout when trying to change state')
        time.sleep(0.001)

    timeout = time.time() + 15
    node.state = 'OPERATION ENABLED'
    while node.state != 'OPERATION ENABLED':
        if time.time() > timeout:
            raise Exception('Timeout when trying to change state')
        time.sleep(0.001)

    #♦print('Node Status {0}'.format(node.powerstate_402.state))

    # -----------------------------------------------------------------------------------------
    node.nmt.start_node_guarding(0.01)
    while True:
        try:
            network.check()
        except Exception:
            break

        # Read a value from TxPDO1
        node.tpdo[1].wait_for_reception()
        speed = node.tpdo[1][0x606C].phys
        modo = node.tpdo[1][0x6061].raw
        # Read the state of the Statusword
        statusword = node.sdo[0x6041].raw

        print('statusword: {0}'.format(statusword))
        print('VEL: {0}'.format(speed))
        print(modo)

        time.sleep(0.01)

except KeyboardInterrupt:
    pass
except Exception as e:
    exc_type, exc_obj, exc_tb = sys.exc_info()
    fname = os.path.split(exc_tb.tb_frame.f_code.co_filename)[1]
    print(exc_type, fname, exc_tb.tb_lineno)
    traceback.print_exc()
finally:
    # Disconnect from CAN bus
    print('going to exit... stopping...')
    if network:

        for node_id in network:
            node = network[node_id]
            node.nmt.state = 'PRE-OPERATIONAL'
            node.nmt.stop_node_guarding()
        network.sync.stop()
        network.disconnect()

Now the dervo goes in run but the engine never move it remain like as braked , he don't have the brake.
Is like the velocity is settled to 0 but I set it to 0x200
do you have some ideas or see something wrong in the file

acolomb, 2022-04-26:
Please try to format code examples as code here in the comments.

You're trying to change the 402 Power State to `SWITCH ON DISABLED` while the node is not yet in the NMT state `OPERATIONAL`.  I think you should better finish the initialization and communication settings first, then go to `OPERATIONAL`, then start manipulating the DS402 state machine.

I don't understand what you're trying to achieve with the manual RPDO transmissions.  The operation mode (0x6060) and the control word are all mapped and handled be the p402 module.  Usually you just start a periodic transmission of the RPDO just like you did for the SYNC service.  Your 2 ms cycle period seems very low actually, have you tried bigger values just in case the controller is too slow for this communication interval?

Do you actually get PDO and SYNC communication on the bus?  What does the failure look like if you say it "doesn't move"?  How far does the script execute properly?

friederschueler, 2024-06-14:
Closing this, as no further response was made.

closed 2024-06-14 by friederschueler