canopen-python/canopen #310: self.bus = can.interface.Bus(*args, **kwargs) AttributeError:

ytuemrah, 2022-04-14:
Hi All,
I have worked with master repository,
I have created an example python file, which includes following codes, in canopen folder:
```
import canopen
network = canopen.Network()
node = canopen.RemoteNode(6, '/home/ubuntu/Documents/bilgim/python/canopen/test/sample.eds')
network.add_node(node)
network.connect()
```
I have received the following error:
```
ubuntu@ubuntu:~/Documents/bilgim/python/canopen$ /usr/local/bin/python3 /home/ubuntu/Documents/bilgim/python/canopen/dnm1.py
Complex data type has an unknown or unsupported data type (40)
Traceback (most recent call last):
  File "/home/ubuntu/Documents/bilgim/python/canopen/dnm1.py", line 13, in <module>
    network.connect()
  File "/home/ubuntu/Documents/bilgim/python/canopen/canopen/network.py", line 112, in connect
    self.bus = can.interface.Bus(*args, **kwargs)
AttributeError: 'NoneType' object has no attribute 'interface'
ubuntu@ubuntu:~/Documents/bilgim/python/canopen$ 
```
Also I have tried to run with python 2 as follows:
```
ubuntu@ubuntu:~/Documents/bilgim/python/canopen$ python dnm1.py 
Traceback (most recent call last):
  File "dnm1.py", line 1, in <module>
    import canopen
  File "/home/ubuntu/Documents/bilgim/python/canopen/canopen/__init__.py", line 1, in <module>
    from .network import Network, NodeScanner
  File "/home/ubuntu/Documents/bilgim/python/canopen/canopen/network.py", line 49
    self.nodes: Dict[int, Union[RemoteNode, LocalNode]] = {}
              ^
SyntaxError: invalid syntax
ubuntu@ubuntu:~/Documents/bilgim/python/canopen$ python3 dnm1.py 
Complex data type has an unknown or unsupported data type (40)
Traceback (most recent call last):
  File "/home/ubuntu/Documents/bilgim/python/canopen/dnm1.py", line 13, in <module>
    network.connect()
  File "/home/ubuntu/Documents/bilgim/python/canopen/canopen/network.py", line 112, in connect
    self.bus = can.interface.Bus(*args, **kwargs)
AttributeError: 'NoneType' object has no attribute 'interface'

```
Also I have tried the following line instead of "network.connect()"
`network.connect(channel='can0', bustype='socketcan')`

I have checked can0 as follows:
```
$ ifconfig can0
can0: flags=193<UP,RUNNING,NOARP>  mtu 72
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
```
$ cat /proc/pcan 

*------------- PEAK-System CAN interfaces (www.peak-system.com) -------------
*------------- Release_20220124_n (8.14.0) Apr 13 2022 08:06:12 --------------
*---------------------- [mod] [isa] [pci] [pec] [usb] -----------------------
*--------------------- 1 interfaces @ major 508 found -----------------------
*n -type- -ndev- --base-- irq --btr- --read-- --write- --irqs-- -errors- status
32    usb   -NA- ffffffff 000 0x031c 00000043 000000a4 03189e25 03189db3 0x000c
```
```
$ lsmod |grep can
can_raw                20480  1
vcan                   16384  0
can_dev                36864  0
pcan                  184320  2
can                    20480  1 can_raw
i2c_algo_bit           16384  3 pcan,i915,nouveau
```
Also I have tried virtual can, can0 as vcan,

Could you guide me at that point?
I have worked on ubuntu and the version is 20.04
How can I run the sample code, how can I establish canopen network connection?

Thank you for your time and consideration,
Best Regards
Emrah

ytuemrah, 2022-04-21:
I think it is about , file path but i am not sure. I have used os module of python to indicate file path, it works for now, i will sure at the end.

closed 2022-04-21 by ytuemrah

ytuemrah, 2023-06-15:
Dear, It was long time ago, I think it is caused of pip or pip3 instalaltion. I mean you maybe install python packets with pip3 instead of pip. Have a good chance.

spaceKelan, 2023-06-15:
I tried that already before, bummer you cannot remember - I wonder if it has to do with the problems described in [my Q&A](https://github.com/christiansandberg/canopen/discussions/382#discussioncomment-6180521)