canopen-python/canopen #317: send and receive data via PDO could not be set up

ytuemrah, 2022-05-13:
Hi All,
I could not set up PDO send and receive mechanism I am gonna share what all code is,
I have used embedded linux platform which include following codes and I have ubuntu host and try to send and receive data, I have managed to send and receive date via SDO but not PDO, could you help at that point?
the embedded side code is as follows , it is a NMT slave and SDO server
`
import signal
import canopen
running = True
def sigint_handler(signum, frame):
        global running
        print('\n\n')
        running = False
        exit(0)
signal.signal(signal.SIGINT,  sigint_handler)
signal.signal(signal.SIGHUP,  sigint_handler)
signal.signal(signal.SIGTERM, sigint_handler)
network = canopen.Network()
network.connect(bustype='socketcan', channel='can0')
node = network.create_node(10, 'e35.eds')
node.nmt.send_command(0)
while running:
        pass
`

the ubuntu side code is as follows, it is a NMT master and SDO client
`
import time
import canopen
import logging
logging.basicConfig(level=logging.DEBUG)
network = canopen.Network()
node = canopen.RemoteNode(10, 'e35.eds')
network.add_node(node)
network.connect(bustype='socketcan', channel='can0')
device_name = node.sdo['Device name'].raw
vendor_id = node.sdo[0x1018][1].raw
# Read PDO configuration from node
node.tpdo.read()
node.rpdo.read()
# Re-map TPDO[1]
node.tpdo[1].clear()
node.tpdo[1].add_variable('Statusword')
node.tpdo[1].add_variable('Velocity actual value')
node.tpdo[1].trans_type = 254
node.tpdo[1].event_timer = 10
node.tpdo[1].enabled = True
node.tpdo[1].save()
# Change state to operational (NMT start)
node.nmt.state = 'OPERATIONAL'
# Read a value from TPDO[1]
node.tpdo[1].wait_for_reception()
speed = node.tpdo[1]['Velocity actual value'].phys
name = node.sdo['Device name']
print("Device name: '{}'".format(name.raw))
`
What is the missing point in the above codes,
the SDO messages will be shown, the SDO messages sent and received via 60A - 58A COB-IDs
there is no 18A, COB-ID.
I have used e35.eds file in the local and remote side,
Thank you for your time and consideration,
Best Regards

ytuemrah, 2022-05-13:
in the debug session in the ubuntu side the code waits in the following point,
`node.tpdo[1].wait_for_reception()`

prunith, 2022-05-13:
Hi,
Perhaps i m wrong but i don't see any command rxpdo1 on the embedded side?

ytuemrah, 2022-05-13:
Dear @prunith ,
Thank you for your reply I have tried something as follows but it did not work, 

```
import signal
import canopen
running = True
def sigint_handler(signum, frame):
	global running
	print('\naber\n')
	running = False
	exit(0)
signal.signal(signal.SIGINT,  sigint_handler)
signal.signal(signal.SIGHUP,  sigint_handler)
signal.signal(signal.SIGTERM, sigint_handler)
network = canopen.Network()
network.connect(bustype='socketcan', channel='can0')
node = network.create_node(10, 'e35.eds')
node.nmt.send_command(0)

# Read a variable using SDO
device_name = node.sdo['Device name'].raw
vendor_id = node.sdo[0x1018][1].raw

# Write a variable using SDO
# node.sdo['Producer Heartbeat Time'].raw = 1000

# Read PDO configuration from node
node.tpdo.read()
node.rpdo.read()
# Re-map TPDO[1]
node.rpdo[1].clear()
node.rpdo[1].add_variable('Statusword')
node.rpdo[1].add_variable('Velocity actual value')
node.rpdo[1].enabled = True
# Save new PDO configuration to node
node.rpdo.save()

# Change state to operational (NMT start)
node.nmt.state = 'OPERATIONAL'

# Read a value from TPDO[1]
node.rpdo[1].wait_for_reception()
speed = node.rpdo[1]['Velocity actual value'].phys
```

prunith, 2022-05-13:
Ubuntu side code was good.

but i think you miss the following step in you linux embedded device:

```
# Start RPDO1 with an interval of 100 ms
node.rpdo[1]['Application Commands.Command Speed'].phys = 1000
node.rpdo[1].start(0.1)
node.nmt.state = 'OPERATIONAL'
```

ytuemrah, 2022-05-13:
Dear @prunith,
I have run following program in the embedded side,
```
import signal
import canopen
running = True
def sigint_handler(signum, frame):
	global running
	print('\naber\n')
	running = False
	exit(0)
signal.signal(signal.SIGINT,  sigint_handler)
signal.signal(signal.SIGHUP,  sigint_handler)
signal.signal(signal.SIGTERM, sigint_handler)

network = canopen.Network()
network.connect(bustype='socketcan', channel='can0')
node = network.create_node(10, 'e35.eds')

node.nmt.send_command(0)

# Read a variable using SDO
device_name = node.sdo['Device name'].raw
vendor_id = node.sdo[0x1018][1].raw

# Write a variable using SDO
# node.sdo['Producer Heartbeat Time'].raw = 1000

# Read PDO configuration from node
node.tpdo.read()
node.rpdo.read()
# Re-map TPDO[1]
node.rpdo[1].clear()
node.rpdo[1].add_variable('Statusword')
node.rpdo[1].add_variable('Velocity actual value')
node.rpdo[1].enabled = True

# Save new configuration (node must be in pre-operational)
node.nmt.state = 'PRE-OPERATIONAL'
node.rpdo.save()

# Change state to operational (NMT start)
# Start RPDO4 with an interval of 100 ms
node.rpdo[1]['Velocity actual value'].phys = 1000
node.rpdo[1].start(0.1)
node.nmt.state = 'OPERATIONAL'
while running:
	pass
```

with the following command,
`node.rpdo[1].start(0.1)`
the candump prints as follows more times:
`(000.100043)  can0  220   [6]  00 00 E8 03 00 00         '......'`
 in the ubuntu side the code wait on the same following point,
`node.tpdo[1].wait_for_reception()`
and there is no any other canopen message in the candump logs

af-silva, 2022-05-13:
Hi @ytuemrah,

I'm gonna try my best to help you. 

`can0 220 [6] 00 00 E8 03 00 00`
220    -> here you have 200 (rpdo1) + NodeID, but the Node ID should be 0A and not 20 (20A -> 0x0A = 10)
00 00 -> Status Word is empty
E8 03 -> Velocity actual value 0x03E8 = 1000
00 00 -> Remaining bytes 

If I'm not mistaken when creating the `network.create_node(nodeid, "file")` nodeid is decimal and so, when using candump you should be seeing 20A and not 220... 

I think the problem is in the EDS file, I don't know at this point why the your nodeid is not being set, it seems it is using the one provided by the eds file.
```
[1400sub1]
ParameterName=COB-Id used
ObjectType=0x7
DataType=0x0007
AccessType=rw
DefaultValue=$NODEID+0x200
ParameterValue=0x220
PDOMapping=0x0
```
Try to things please 
- First remove the ParameterValue please and test;
- Second change the ParameterValue to 0x20A and test;

Additional try to do this.
```python
while True:
  try:
      network.check()
  except Exception:
      break

  time.sleep(0.01)
```

ytuemrah, 2022-05-17:
Dear @af-silva ,
Thank you really much for your time and consideration, 0x220 is a rpdo message and it must be 0x200 + nodeid but the ParameterVaule in the .eds file breaks it so when I have removed ParameterValue from the .eds file, as you said it turns to 0x20A,
Thank you again
Best Regards

closed 2022-05-17 by ytuemrah