canopen-python/canopen #325: Full queue when reusing bus

gardenvariety42, 2022-08-12:
I'm building a program to automatically test a safety control unit for industrial machines. Since the actuators and sensors will be imitated by canopen nodes, I'm forced to make j1939 and canopen play nice with each other. I'm required to have both using the same bus so I don't have to use two peak devices. This is what I've tried:

connection = Connection()
connection.connect(1)
connection.get_messages()

network = Network()
network.bus = connection.ecu._bus
listeners = network.listeners + connection.ecu._listeners
network.notifier = can.Notifier(connection.ecu._bus, listeners, 0.5)

trying to run this I get:
Exception:  Failed to send: The transmit queue is full

I'm a bloody beginner so likely I'm doing something wrong but I just can't figure it out. Any suggestions would be greatly appreciated.

christiansandberg, 2022-08-12:
Do you have some other CAN device physically connected to the bus? A CAN message can only be flushed from the buffers when another node has acknowledged it.

gardenvariety42, 2022-08-15:
Yes, there are nodes physically connected via a Peak USB contoller. I've been working with both, the j1939 node and the CANopen node successfully for a while now. If I have both running  on separate busses via two controllers, everything is fine. The problem only occurs when I try to use the same bus.