canopen-python/canopen #326: Amount of RPDO received in specific COBID in callback is much less than showing on CAN bus

wilson455039, 2022-09-13:
Hi,
We are trying to use python canopen on our project, the task we have to do is configure a node to receive specific COBID's data from other existing hardware. These existing PDO are transmitting messages each sync, and between each sync is 10ms. 

The problem is we did not receive all the messages in a period. For example, for COB-ID 0x281, we saw 2751 on Candump but only received 1141 in the python code. We narrow down the possible issue and found that message might be dropped by the callback function in canopen package.  So we test like:

`def count_pdo(message):`
`  num_pdo_received+=1`

`node.tpdo[1].add_callback(count_pdo)`

And we just get 1141 for `num_pdo_received` and on Candump there are 2751 messages in this COB-ID. Any idea that might cause the issue? Thank you.

christiansandberg, 2022-09-18:
Could you provide more info on the CAN hardware, OS, Python version et.c. It could just be the case that Python is too slow to process all messages on time.

wilson455039, 2022-09-20:
We also suspect that might be the case. We're running this on Nvidia Jetson Xavier on Ubuntu 18.04 with a USB to can adaptor and python 3.6.9.
We see every message on candump just fine. However, when are running using the canopen library, we see that even on the notifier class layer - it is not receiving all messages.
Any idea on how to optimise this?

af-silva, 2022-10-04:
Hi @wilson455039 

Like christiansandberg mention, python is not as fast as compiled languages, and in this case your problem may laid there. One thing you can try is to run the code using PyPy instead of Python runtime, and see if you can get it to work, I think it should work but I can't test it myself.

Best of lucks, and if you get it to run please provide feedback if you can.

Thanks

friederschueler, 2024-06-14:
Closing this for now, as the problem is very old and under the notifier class layer, which is the python-can and/or OS layer.

closed 2024-06-14 by friederschueler