canopen-python/canopen #328: LocalNode not emitting TPDOs

nick-wlink, 2022-09-20:
Great library. Generally it has been nice to use, but I'm having some trouble with PDOs.

I would like to use the canopen library to simulate a number of slave devices. Configuration of the slaves will be read from an eds/dcf file and thereafter the bus master may make further configuration changes to the slave devices PDOs. In the final application the bus master will be a separate physical device (_not_ the canopen library).

To begin I am trying to set up a single slave and get it to transmit a PDO. For convenience I am using the canopen library to also simulate my bus master. I have two python repls open. Top left is the bus master, bottom left is the slave and right is a dump of the virtual CAN bus they are communicating over.

My expectation was that a slave with event-driven asynchronous TPDOs configured would start transmitting the TPDO as soon as it was put into 'OPERATIONAL' mode. I'm new to CANOpen, so I'm not sure if this expectation is correct. But this is the case with an actual CANOpen sensor (encoder) that I've been experimenting with. You can see in the screen capture that I've tried to configure the slave PDO (most of the config and mappings are already done in the DCF file) and put it into operational mode, but I don't get any PDOs in the candump.

[image]

I notice that there is a `tpdo.start(interval)` function in the API, but I don't quite understand how this works. Is this supposed to be called on the device that is emitting the TPDO (the slave device in my case)? How do I use this if I don't know in advance which PDOs the master will configure? Is there some kind of callback that I should invoke it from when the slave is put into operational mode?

Also, when I call the `start()` function I am getting an error:
```
>>> node.tpdo[1].start(1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/nick/.local/lib/python3.10/site-packages/canopen/pdo/base.py", line 498, in start
    self._task = self.pdo_node.network.send_periodic(
  File "/home/nick/.local/lib/python3.10/site-packages/canopen/network.py", line 233, in send_periodic
    return PeriodicMessageTask(can_id, data, period, self.bus, remote)
  File "/home/nick/.local/lib/python3.10/site-packages/canopen/network.py", line 311, in __init__
    self.msg = can.Message(is_extended_id=can_id > 0x7FF,
TypeError: '>' not supported between instances of 'NoneType' and 'int'
```

windelbouwman, 2022-09-29:
Exact same issue here. When searching for TPDO in the issues, I found this duplicate issue: #166

af-silva, 2022-10-04:
Hi @nick-wlink and @windelbouwman the Local Node has a limited implementation. If you like and want to tinker with the code, you can try to enable that functionality of sending TPDO from Local Nodes and submit a PR to extend the library.

base-jumper, 2022-10-06:
Thanks @windelbouwman and @af-silva for the explanation.

closed 2024-06-13 by friederschueler