canopen-python/canopen #339: Project state inquiry

slightlynybbled, 2022-12-22:
Greetings -

I noticed that the project appears to have recent commits, but that there are several open pull requests from some time ago.  In getting started on the project, there were some headwinds, particularly within documentation and type-hints.

 - Are you accepting pull requests?  If so, why are there open PRs from a while back?
 - Are you interested in having type hints?  I don't want to contribute in this area if you aren't going to use them.
 - Are you interested in improved documentation?

Thanks, and nice project!

j

acolomb, 2022-12-22:
Hi and welcome to the project. Contributions are definitely welcome. Development is not really fast paced right now, but usually new features get added when someone has an itch to scratch in their project. I guess we need to take a look at open PRs again, I don't have an overview what's currently left to review. But as far as I know nothing urgent, otherwise we'd have looked at it. Some things might just be leftovers that have been superseded by a different solution.

I have some improvements in mind actually that I want to add sometime, but usually the bottleneck is time for testing changes on real hardware.

Regarding type hints, I'd welcome having them. Some of the newer code includes them, but definitely not consistently. That will automatically help documentation as well, and from my experience we'd need  some more examples too, because some API parts can only be understood by looking at the code (which is also always a good thing to do btw.)

To sum it up, the project is mildly active and open to contributions if you feel like making some PRs.

friederschueler, 2024-06-13:
Question is answered and in recent times the project gained activity. Also we implemented type hints.

closed 2024-06-13 by friederschueler