canopen-python/canopen #340: Calling add_node multiple times & network.subscribe issue

samsamfire, 2023-01-02:
Hello,

I've encountered this bug multiple times but have finally got down to figuring out what was going on but basically : 
- network.add_node creates a new remote node object and sets the key inside dictionary accordingly by calling the method "associate_network" inside remote node which calls the network.subscribe for specific CANopen frames.

However if you look at the subscribe method :

```
def subscribe(self, can_id: int, callback: Callback) -> None:
        """Listen for messages with a specific CAN ID.

        :param can_id:
            The CAN ID to listen for.
        :param callback:
            Function to call when message is received.
        """
        self.subscribers.setdefault(can_id, list())
        if callback not in self.subscribers[can_id]:
            self.subscribers[can_id].append(callback)
```
It checks if callback is not already present in the subscribers callback list. This means that if add_node was already called for a specific id, the subscribers list will not be updated accordingly and will keep the old callbacks from the previous RemoteNode object.

I see two options for fixing this:
- Changing the checks made on the subscribe method
- Calling __setitem__ should check if id was already present in list and call remove_network before creating new node.

Calling add_node multiple times can be useful in case the actual physical device changes for the same id.

What are your thoughts on this ?

Thanks !

acolomb, 2023-01-02:
I think this is a very niche use case, when would you exchange a device and re-assign the same device ID, all while your software keeps running? I only see how that could be useful for a diagnosis tool or some software to configure nodes manually. In such a setting, why not simply remove the node before adding it again, from your code?

samsamfire, 2023-01-02:
I have a use case where I want to update a device and the actual EDS on the actual device does change.
Anyway, yes previously this is what I did because I didn't know what was causing this problem.
In any case I think that even though you might think that it is a niche use case, it still is a bug because the second time you call the add_node function, the remote node inside the network dictionary does not have the proper callbacks implemented, and all of this is done silently.
Personally I don't see anything wrong with re-assigning a value to a key inside a dictionary so if this is not possible there should at least be an exception. (There is no reason to call del when updating a dictionary key-value pair)

acolomb, 2023-01-05:
I agree that failing silently in this way is wrong, leaving everything in a non-working state. Care to make a PR to either support the node update or add an exception?

Do we have any precedence in the code what `add_something()` will do when the thing is already present? I'd think the principle of least surprise would tell us that it should be replaced and work afterwards.

samsamfire, 2023-01-05:
Well we have functions like "add_callback" that just append callbacks to a list, but here it doesn't make sense to have multiple nodes with the same ID, so I think we should just overwrite the previous node.
I just thought over another use case I had for calling add_node multiple times : I have nodes with certain versions and if I don't know beforehand what EDS is on the device, I need to first add it, then get the software version and retreive the correct EDS.

Anyway, I'll make a PR, thanks ;)

closed 2023-01-17 by samsamfire