canopen-python/canopen #346: Drop Python 2.7 support for better code quality and support of new language features

friederschueler, 2023-02-22:
Hi,

I would like to propose to drop the support for python 2.7. Official support for this version (EOL) ended on 31.12.2020 which was three years ago and over 12 years after 2.7 was released and over 20 years after Python 2.x. Even Debian oldstable has support for Python 3.7.
My main reason for this suggestion is code quality and the new language features (like type hints!). My PyLint can't handle the mixed style of the current scripts and things like:

`try:
    from configparser import RawConfigParser, NoOptionError, NoSectionError
except ImportError:
    from ConfigParser import RawConfigParser, NoOptionError, NoSectionError

    try:
        # Python 3
        eds.read_file(fp)
    except AttributeError:
        # Python 2
        eds.readfp(fp)
`

 are ugly and smelly and can lead to further code errors. What do you guys think?

acolomb, 2023-02-22:
I'd welcome such an endeavor.

Someone "just" needs to volunteer for the work involved, including testing that those changes don't alter existing behavior. Which is kind of the current road-block I guess, things that get touched and tested will not - try to stay compatible and thus Python2 support will bit-rot slowly. But testing code you're not currently involved with anyway is really hard and thus not likely to happen unless it scratches someone's itch.

friederschueler, 2023-02-22:
>  But testing code you're not currently involved with anyway is really hard and thus not likely to happen unless it scratches someone's itch.

Well coverage from unit tests is not too bad.

```
Name                                    Stmts   Miss  Cover   Missing
---------------------------------------------------------------------
canopen\__init__.py                        11      2    82%   8-10
canopen\_version.py                         2      0   100%
canopen\emcy.py                            71     19    73%   37, 46, 50-51, 63-79, 93-94
canopen\lss.py                            185    109    41%   6-7, 100-104, 108, 133-142, 152, 165, 173, 186, 197-201, 206, 231-236, 240-242, 254-283, 286-297, 300-309, 317-326, 334-343, 347-360, 374-396, 399
canopen\network.py                        168     36    79%   3-4, 13-17, 82-85, 133, 136, 161-162, 207, 263-264, 275-276, 279, 330-338, 353, 357-359, 393, 397-401
canopen\nmt.py                            132     22    83%   94, 101, 126, 152, 157-166, 174, 182-183, 187-189, 232, 260
canopen\node\__init__.py                    2      0   100%
canopen\node\base.py                        9      0   100%
canopen\node\local.py                      80     10    88%   47-54, 68, 101
canopen\node\remote.py                     80     34    58%   51, 67-77, 107, 118, 127-150, 154-161
canopen\objectdictionary\__init__.py      280     47    83%   8-9, 33-41, 65, 71, 79-83, 113-114, 117-119, 152, 184-185, 188-190, 250, 366, 371-376, 380, 385, 391, 394, 400-405, 422, 424, 431, 436-438
canopen\objectdictionary\datatypes.py      21      0   100%
canopen\objectdictionary\eds.py           326     41    87%   9-10, 33-35, 53, 112-116, 178-192, 199, 205, 234-243, 292-293, 306-307, 337, 357, 364, 396-397, 432, 494-495
canopen\pdo\__init__.py                    35      8    77%   48-52, 73-77
canopen\pdo\base.py                       354    108    69%   6-7, 34, 48, 51, 71-72, 83-119, 222, 233, 236, 241, 248-254, 270-275, 284-292, 295-305, 314, 333, 340, 347, 357-359, 374-375, 377-378, 380-381, 386-391, 394-404, 407-414,
 455, 472, 477, 489-498, 504-505, 510, 516-517, 525-528
canopen\profiles\__init__.py                0      0   100%
canopen\profiles\p402.py                  275    176    36%   99-101, 212-215, 224-227, 238-244, 247-254, 259-264, 267-268, 273-274, 279-284, 290-300, 303-304, 309-315, 324-331, 342-369, 394-405, 409-431, 435-437, 449-455, 463-464, 
473-477, 491-499, 508, 512-518, 541-545, 549-556, 559-568, 571-581
canopen\sdo\__init__.py                     4      0   100%
canopen\sdo\base.py                        75     13    83%   5-6, 64, 67, 70, 79, 92, 95, 98, 108, 111, 114, 117
canopen\sdo\client.py                     453     87    81%   7-8, 52, 54-60, 68-69, 80, 87-91, 218, 256, 260, 278, 304, 306, 319-321, 324, 327, 364, 379, 384, 386, 392, 416, 490, 515, 519-520, 527-528, 541-542, 548-560, 569, 575-57
6, 578-579, 585, 600-602, 643, 648-649, 653-654, 675, 683, 731-732, 735-736, 740-742, 752-769, 774, 790
canopen\sdo\constants.py                   27      0   100%
canopen\sdo\exceptions.py                  12      4    67%   52-55
canopen\sdo\server.py                     127      9    93%   50, 53-57, 85, 138, 154
canopen\sync.py                            19      7    63%   32-38, 42-43
canopen\timestamp.py                       15      0   100%
canopen\variable.py                        97     42    57%   5-6, 29, 32, 80, 99-102, 106, 111-113, 117, 122, 138-143, 158-163, 169-170, 174-180, 183, 186-188, 191, 194, 197, 200
test\test_eds.py                          134      0   100%
test\test_emcy.py                          44      0   100%
test\test_local.py                        173      1    99%   269
test\test_network.py                       68     13    81%   63-76, 90
test\test_od.py                           112      0   100%
test\test_pdo.py                           45      1    98%   60
test\test_sdo.py                           72      1    99%   171
test\test_sync.py                          24      1    96%   29
test\test_time.py                          15      1    93%   20
---------------------------------------------------------------------
TOTAL                                    3547    792    78%
```

I would like to contribute some minor fixes and changes, but in the current state my IDE (PyDev+PyLint) reports a lot of errors, which slows down the development a lot for me. I will start a 3.x only branch and then maybe provide a PR if you are interested. 
If I get time, I will also write some more unit tests for better coverage.

christiansandberg, 2023-02-22:
Yes that would be very welcome. We already dropped support a long time ago but haven't refactored existing code.

sveinse, 2024-05-19:
I believe with PR #433 and PR #441, we have done away with the python 2 support and I think this issue may be closed.

PR #441 (2024-05-26): Minor linting nitpicks and Python 2 removal
* Removed minor formatting/linting nitpicks that doesn't change any code
* Removed py2 handling in import_eds
* Made file open more robust in import_eds

This PR is a simple cleanup. It removes some old references to python 2, including the open file logic in `import_eds()`. It closes #346
closed 2024-05-26 by acolomb