canopen-python/canopen #353: Reading PDO without Object Dictionary

danielw1210, 2023-03-22:
Hello all,

Is There possibility to read PDO's from node created without Object Dictionary (.eds)? I've tried multiple ways, but i've got error that

`node.tpdo[1]` cannot be accessed `KeyError: 1`

I would be grateful if you could help me, because I didn't find any information about it.

christiansandberg, 2023-03-24:
No not currently. But it would be a welcome change. The library needs the data types of the indexes it uses but since the PDO configuration is standardized, the reading and parsing could be hard coded.

acolomb, 2023-03-24:
The library relies on the object dictionary quite heavily to provide all its CANopen features. Without it, you won't be using SDO efficiently either. So if it's just decoding PDOs with a fixed structure you're after, you might have better luck using just the lower level python-can and decoding using the struct or bitstruct modules (from the top of my head, could be different names, but definitely in the standard library).

Or just write your own, as-simple-as-possible EDS file to work with.

sveinse, 2023-03-25:
I have a scheme where I'm quering the connected remote node to determine what proper OD to use, and to do that I make a temporary OD:

```python
import canopen

od = canopen.objectdictionary.ObjectDictionary()
var = canopen.objectdictionary.Variable('Device Type', 0x1000, 0)
var.data_type = canopen.objectdictionary.INTEGER32
od.add_object(var)
```

Perhaps something similar can be used to programmatically create the objects needed for the PDOs?

friederschueler, 2024-06-13:
Closing this, because the original question is answered. You can also use https://github.com/CANopenNode/CANopenEditor to create your own, as-simple-as-possible EDS file to work with.

closed 2024-06-13 by friederschueler