canopen-python/canopen #354: timeout boot up error

superlogan72, 2023-03-24:
Hello i have write just few line under linux:
import canopen
import os
import traceback
import time
import os
try:
    
    os.system('sudo ip link set can0 up type can bitrate 250000')
    network = canopen.Network()
    network.connect(bustype="socketcan", channel='can0')
    network.check()
    node = canopen.BaseNode402(3, '/home/pi/Desktop/TBS3.eds')
    network.add_node(node)
    node.nmt.state = 'RESET COMMUNICATION'
    node.nmt.wait_for_bootup(15)
except:
    ..........
#any time i get timeout

friederschueler, 2024-06-13:
Please provide more information. How is the (raw) can-bus looking and was the device working in some other environment? Also is it configured? It seems, that it does not send heartbeats.

acolomb, 2025-09-22:
No response.

closed 2025-09-22 by acolomb