canopen-python/canopen #364: pytest fails on Windows

sveinse, 2023-04-07:
pytest of canopen fails on Windows.

### Failure 1

```
FAIL: TestEDS.test_export_eds
canopen\objectdictionary\eds.py:29: PermissionError
```
Windows doesn't allow reread of `NamedTemporaryFile` as is done in `canopen.import_od()`

```py
          # From test_eds.py
            with tempfile.NamedTemporaryFile(suffix="." + doctype, mode="w+") as tempeds:
                print("exporting %s to " % doctype + tempeds.name)
                canopen.export_od(self.od, tempeds, doc_type=doctype)
                tempeds.flush()
                exported_od = canopen.import_od(tempeds.name)
```

### Failure 2

```
FAIL: TestNetwork.test_send_perodic
test\test_network.py:62: AssertionError
```
Since `send_periodic()` is asyncronous, I think the window for counting items is a bit narrow. It depends on the timing for the machine it runs and empirically its too few on Windows.

samsamfire, 2023-04-07:
I can confirm that I had those problems on windows as well

sveinse, 2023-04-08:
Fixed in PR #365

closed 2023-04-08 by sveinse