canopen-python/canopen #372: Problem in profile positioning

superlogan72, 2023-04-12:
Hello this is my short script, but actually the servo never move, he goes in run ( i can't turn in but it never move), did you have some ideas?
import canopen
import sys
import os
import traceback
import time
import logging
logging.basicConfig(level=logging.DEBUG)

try:

    # Start with creating a network representing one CAN bus
    os.system('sudo ip link set can0 up type can bitrate 500000')
    network = canopen.Network()
    network.connect(bustype="socketcan", channel='can0')

    network.check()

    # Add some nodes with corresponding Object Dictionaries
    node = canopen.BaseNode402(1, '/home/pi/Desktop/TBS3.eds')
    network.add_node(node)
    # Reset network
    node.nmt.wait_for_bootup(15)

    print('node state 1) = {0}'.format(node.nmt.state))
    # Iterate over arrays or records
    error_log = node.sdo[0x1003]
    for error in error_log.values():
        print("Error {0} was found in the log".format(error.raw))

    for node_id in network:
        print(network[node_id])

    print('node state 2) = {0}'.format(node.nmt.state))

    # Transmit SYNC 
    network.sync.start(0.02)

    node.load_configuration()
    print('node state 3) = {0}'.format(node.nmt.state))
    node.tpdo[1].clear()
    node.tpdo[1].add_variable(0x6041)
    node.tpdo[1].add_variable(0x6061)
    node.tpdo[1].add_variable(0x606C)
    
    node.tpdo[1].trans_type = 1
    node.tpdo[1].enabled = True
    node.tpdo.save()
    node.rpdo[1].clear()
    node.rpdo[1].add_variable(0x6040)
    node.rpdo[1].add_variable(0x6060)
    node.rpdo[1].add_variable(0x60FF)
    node.rpdo[1].enabled = True
    node.rpdo.save()
    time.sleep(1)

    node.setup_402_state_machine()

    device_name = node.sdo[0x1008].raw
    vendor_id = node.sdo[0x1018][1].raw

    print(device_name)
    print(vendor_id)
    node.sdo[0x6085].raw=2500000
    node.sdo[0x6060].raw=1
    node.sdo[0x6040].raw= 0x06
    node.sdo[0x6040].raw= 0x0F
    node.sdo[0x607A].raw= 100000
    node.sdo[0x6081].raw=58333
    node.sdo[0x6083].raw=1000000
[... 547 lines cut ...]
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 02 16 00 00 00 00 00
DEBUG:canopen.variable:Writing RPDO 3 Communication Parameter.COB-ID used by PDO (0x1402:1) = 1025
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    23 02 14 01 01 04 00 00
INFO:canopen.pdo.base:Subscribing to enabled PDO 0x401 on the network
INFO:canopen.pdo.base:Setting COB-ID 0x501 and temporarily disabling PDO
DEBUG:canopen.variable:Writing RPDO 4 Communication Parameter.COB-ID used by PDO (0x1403:1) = 2147484929
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    23 03 14 01 01 05 00 80
INFO:canopen.pdo.base:Setting transmission type to 0
DEBUG:canopen.variable:Writing RPDO 4 Communication Parameter.Transmission Type (0x1403:2) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 03 14 02 00 00 00 00
DEBUG:canopen.variable:Writing RPDO 4 Mapping Parameter.Number of mapped objects (0x1603:0) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 03 16 00 00 00 00 00
DEBUG:canopen.variable:Writing RPDO 4 Mapping Parameter.Number of mapped objects (0x1603:0) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 03 16 00 00 00 00 00
INFO:canopen.pdo.base:Setting COB-ID 0x0 and temporarily disabling PDO
DEBUG:canopen.variable:Writing RPDO 5 Communication Parameter.COB-ID used by PDO (0x1404:1) = 2147483648
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    23 04 14 01 00 00 00 80
I

superlogan72, 2023-04-12:
debug part 2:
NFO:canopen.pdo.base:Setting transmission type to 0
DEBUG:canopen.variable:Writing RPDO 5 Communication Parameter.Transmission Type (0x1404:2) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 04 14 02 00 00 00 00
DEBUG:canopen.variable:Writing RPDO 5 Mapping Parameter.Number of mapped objects (0x1604:0) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 04 16 00 00 00 00 00
DEBUG:canopen.variable:Writing RPDO 5 Mapping Parameter.Number of mapped objects (0x1604:0) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 04 16 00 00 00 00 00
INFO:canopen.pdo.base:Setting COB-ID 0x0 and temporarily disabling PDO
DEBUG:canopen.variable:Writing RPDO 6 Communication Parameter.COB-ID used by PDO (0x1405:1) = 2147483648
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    23 05 14 01 00 00 00 80
INFO:canopen.pdo.base:Setting transmission type to 0
DEBUG:canopen.variable:Writing RPDO 6 Communication Parameter.Transmission Type (0x1405:2) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 05 14 02 00 00 00 00
DEBUG:canopen.variable:Writing RPDO 6 Mapping Parameter.Number of mapped objects (0x1605:0) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 05 16 00 00 00 00 00
DEBUG:canopen.variable:Writing RPDO 6 Mapping Parameter.Number of mapped objects (0x1605:0) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 05 16 00 00 00 00 00
INFO:canopen.pdo.base:Setting COB-ID 0x0 and temporarily disabling PDO
DEBUG:canopen.variable:Writing RPDO 7 Communication Parameter.COB-ID used by PDO (0x1406:1) = 2147483648
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    23 06 14 01 00 00 00 80
INFO:canopen.pdo.base:Setting transmission type to 0
DEBUG:canopen.variable:Writing RPDO 7 Communication Parameter.Transmission Type (0x1406:2) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 06 14 02 00 00 00 00
DEBUG:canopen.variable:Writing RPDO 7 Mapping Parameter.Number of mapped objects (0x1606:0) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 06 16 00 00 00 00 00
DEBUG:canopen.variable:Writing RPDO 7 Mapping Parameter.Number of mapped objects (0x1606:0) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 06 16 00 00 00 00 00
INFO:canopen.pdo.base:Setting COB-ID 0x0 and temporarily disabling PDO
DEBUG:canopen.variable:Writing RPDO 8 Communication Parameter.COB-ID used by PDO (0x1407:1) = 2147483648
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    23 07 14 01 00 00 00 80
INFO:canopen.pdo.base:Setting transmission type to 0
DEBUG:canopen.variable:Writing RPDO 8 Communication Parameter.Transmission Type (0x1407:2) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 07 14 02 00 00 00 00
DEBUG:canopen.variable:Writing TPDO 8 Mapping Parameter.Number of mapped objects (0x1607:0) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0601    S Rx                DL:  8    2f 07 16 00 00 00 00 00
DEBUG:canopen.variable:Writing TPDO 8 Mapping Parameter.Number of mapped objects (0x1607:0) = 0
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
[... cut ...]
INFO:canopen.nmt:Changing NMT state on node 1 from PRE-OPERATIONAL to PRE-OPERATIONAL
INFO:canopen.nmt:Sending NMT command 0x80 to node 1
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0000    S Rx                DL:  2    80 01
DEBUG:can.interfaces.socketcan.socketcan:Stopping periodic task
DEBUG:can.interfaces.socketcan.socketcan:Closing bcm socket for channel can0
DEBUG:can.interfaces.socketcan.socketcan:Closing raw can socket

acolomb, 2023-04-12:
That is a lot of info to parse, please cut it down to what, you see that's going wrong.

Some general thoughts: Are you setting the device to OPERATIONAL state after configuring PDOs? Have you started the PDO transmission?

You seem to be using only a very small part of the DS402 profile API. There are properties implemented on the node class to switch e.g. the operation mode and the DS402 power state machine. You should probably look at the relevant docs again or the p402.py file to understand the intended usage. Fiddling around with object index numbers in hex shouldn't be necessary at all, that's what this library tries to abstract away.

superlogan72, 2023-04-14:
Hello again, sorry but i'm old (74 y.o.)
I start from the beginning.
import canopen
from canopen.profiles.p402 import BaseNode402
import os
from time import sleep
os.system('sudo ip link set can0 up type can bitrate 500000')
network = canopen.Network()
network.connect(bustype="socketcan", channel='can0')
node = BaseNode402(1, '/home/pi/Desktop/TBS3.eds')
network.add_node(node)
network.check()
network.sync.start(0.02)
node.load_configuration()
sleep(1)
node.nmt.state = 'RESET COMMUNICATION'
node.nmt.wait_for_bootup(15)
node.load_configuration()
node.tpdo[1].clear()
node.tpdo[1].add_variable(0x6041)
node.tpdo[1].add_variable(0x6061)
node.tpdo[1].add_variable(0x606C)
node.tpdo[1].trans_type = 1
node.tpdo[1].enabled = True
node.tpdo.save()
node.rpdo[1].clear()
node.rpdo[1].add_variable(0x6040)
node.rpdo[1].add_variable(0x6060)
node.rpdo[1].add_variable(0x60FF)
node.rpdo[1].enabled = True
node.rpdo.save()
node.setup_402_state_machine()
sleep(1)
print('stato',node.state)
**#i get not ready to switch on Why??  what is wrong**
if i commentt out the following:
node.tpdo[1].clear()
node.tpdo[1].add_variable(0x6041)
node.tpdo[1].add_variable(0x6061)
node.tpdo[1].add_variable(0x606C)
node.tpdo[1].trans_type = 1
node.tpdo[1].enabled = True
node.tpdo.save()
node.rpdo[1].clear()
node.rpdo[1].add_variable(0x6040)
node.rpdo[1].add_variable(0x6060)
node.rpdo[1].add_variable(0x60FF)
node.rpdo[1].enabled = True
node.rpdo.save()
i get:
Controlword not configured in node 1's PDOs. Using SDOs can cause slow performance.
Statusword not configured in node 1's PDOs. Using SDOs can cause slow performance.
Operation Mode not configured in node 1's PDOs. Using SDOs can cause slow performance.
Operation Mode Display not configured in node 1's PDOs. Using SDOs can cause slow performance.
The object 0x6041 is not a configured TPDO, fallback to SDO
The object 0x6041 is not a configured TPDO, fallback to SDO
The object 0x6041 is not a configured TPDO, fallback to SDO
But the state (ready to switch on)

superlogan72, 2023-04-14:
DEBUG:can.interfaces.socketcan.socketcan:Reading properties of (cyclic) transmission task id=1
DEBUG:can.interfaces.socketcan.socketcan:Invalid argument - transmission task not known to kernel
Can you help me to uderstad this?