canopen-python/canopen #373: Problem to understand tpdo

superlogan72, 2023-04-20:
Sorry if i ask again:
network = canopen.Network()
network.connect(bustype="socketcan", channel='can0',bitrate=1000000)
network.check()
node = BaseNode402(1, '/home/pi/Desktop/file.eds')
network.add_node(node)
network.sync.start(0.02)
sleep(1)
node.nmt.state = 'RESET'
node.nmt.wait_for_bootup(15)
node.load_configuration()
___Till here is all ok--
than as i undrstand i have to load the tpdo so i write at first:
node.tpdo[1].clear()
node.tpdo[1].add_variable(0x6041)
node.tpdo[1].trans_type = 1
node.tpdo[1].enabled = True
node.tpdo.save()
on the save i get this error:
Value 2147484033 is greater than max value 2047 That i don't understand

if i never load the tpdo i get 
The object 0x6041 is not a configured TPDO, fallback to SDO
what i'm doing wrong
Thanks

acolomb, 2023-05-02:
Can you show the backtrace of the error?  What exactly is it trying to write via SDO when it fails?

acolomb, 2024-06-11:
Closing after a year without feedback.

closed 2024-06-11 by acolomb