canopen-python/canopen #377: Index + Subindex missing from abort messages

mbs38, 2023-05-12:
Hi, thanks for the great work.

I noticed that index + subindex are missing from the client side abort messages when comparing the raw data to a commercial tool I'm using. Index + subindex have to be included according to the standard. See https://forum.opencyphal.org/uploads/short-url/mNWuvY23DYckSFoSbfam0ji3YWn.pdf Page 61.

Code: 
https://github.com/christiansandberg/canopen/blob/e8807b87e69902840f459279ce38e330b069f53d/canopen/sdo/client.py#LL92C1-L100C1

Thx!

friederschueler, 2024-06-13:
I think this is a bug, because we are not following the standard.

Could you provide the different raw data?

dhairyajangir, 2026-11-11:
Here is an independent raw-frame comparison for the requested data. I reproduced the missing address using two actual `python-can` VirtualBus endpoints with a CANopen notifier and a scripted peer. These are client frames on COB-ID `0x602`, DLC 8, for transfers to `0x2345:67`:

| Trigger | Current master (`14b3336`) | Updated #686 |
| --- | --- | --- |
| Initiation timeout | `80 00 00 00 00 00 04 05` | `80 45 23 67 00 00 04 05` |
| Segmented upload, toggle mismatch | `80 00 00 00 00 00 03 05` | `80 45 23 67 00 00 03 05` |
| Segmented download, invalid response | `80 00 00 00 01 00 04 05` | `80 45 23 67 01 00 04 05` |

Only address bytes 1–3 change in these comparisons; the public exceptions, abort reasons and COB IDs remain the same. The updated client stores each stream's requested address, including for later segment and timeout paths. The same four VirtualBus cases, including the separate block-upload end-timeout regression, passed on Python 3.9 and 3.12.

This is not a comparison against the reporter's commercial tool, physical hardware, or a peer that rejects zero-address aborts. Direct `abort()` calls still need explicit address keywords to send a nonzero address.

AI assisted with the investigation, patch and tests.