canopen-python/canopen #388: Unknown interface type when using "iscan" parameter for bustype

NixonLuka, 2023-08-01:
When using the canopen module with an iscan interface I get a non implemented error, despite this interface being implemented in python-can. Is there a way to fix this?

To reproduce this I simply write:

```
network = canopen.Network()
bus = can.ThreadSafeBus(bustype = "iscan", channel = 0, bitrate = 500000)
network.connect(bus)
```

Throws: 
`raise CanInterfaceNotImplementedError(can.exceptions.CanInterfaceNotImplementedError: Unknown interface type "None"`

friederschueler, 2024-06-13:
The correct parameter name is "interface" not "bustype" anymore.

acolomb, 2024-06-13:
In addition, if you want to pass a pre-initialized `BusABC` object to `canopen.Network`, you need to do that in the constructor, not as an argument to `network.connect()`.

    bus = can.ThreadSafeBus(bustype = "socketcan", channel = "vcan0", bitrate = 500000)
    network = canopen.Network(bus)
    network.connect()

closed 2024-06-13 by acolomb