canopen-python/canopen #391: Does the dictionary in the EDS file have any impact on uppercase or lowercase names?

zouzhe1, 2023-08-19:
I don't think it's appropriate to find a corresponding dictionary by name, 
because different manufacturers' dictionaries have different names, such as uppercase and lowercase。
For example, one of my devices is **['actual speed ']**, but your routine is **['Velocity actual value']**

I think you should use the cia standard to represent it, such as 6040，1008，6060 and so on, in numerical form.

Also, using strings to represent NMT status is not safe, although it may seem intuitive. Due to the possibility of accidentally typing another character, the NMT status may be incorrect. For example, node. state='READY TO SWITCH ON', this string is easy to write as 'READY TO     SWITCH ON'

acolomb, 2023-08-19:
So what exactly are you proposing to change?

zouzhe1, 2023-08-19:
@acolomb Using variables or macro definitions, or enumerations, to replace strings

acolomb, 2023-08-19:
Okay, but where exactly?

Macros don't exist in python as they do in C for example. Enumerations neither. In Python, every symbol is just a pointer to an object. Any object has a hash, which is used when looking up keys in a map.

friederschueler, 2024-06-13:
Also you could jsut use the hex values to access sdo if you don't wanna use strings:

node.sdo[0x1000].raw