canopen-python/canopen #392: TPDO 2 Subindex

fab-sch, 2023-08-28:
Hi there,
i mapped 0x3010sub1, 0x3010 sub2, and 0x3010 sub3 to TPDO2. 
No i want to read the 0x3010sub1 with a python script:
...
    try:
        while True:

            node_50.tpdo[2].wait_for_reception()
            x_raw=node_50.tpdo[2][0x3010][1].raw 

            print("IMU_Ges_x:", x_raw)

    except KeyboardInterrupt:
        pass
...

unfortunately i always get the error:     
x_raw=node_50.tpdo[2][0x3010][1].raw 
TypeError: 'Variable' object is not subscriptable

I also tried x_raw=node_50.tpdo[2][0x3010, 1].raw but the same error occured

When I try x_raw=node_50.tpdo[2][0x3010].raw i recieve values, but how can i access to 3010sub3 for example? 
Thank you very much

af-silva, 2023-09-05:
Hi @fab-sch,

If you mapped the objects you can use the parameter name to access it 

```
[1600sub1]
ParameterName=Target Velocity
ObjectType=0x7
DataType=0x0007
AccessType=rw
DefaultValue=0x60ff0020
PDOMapping=0x0
ParameterValue=0x60ff0020
```

For instance "Target Velocity", is the name of the object mapped to RPDO 1 subindex 1
If you want to access the subindex of a object use the "." notation like:

try this (this is from the top of my head and i'm not sure which is the correct one)
`val = node.tpdo[1][0x607D.607Dsub1].raw`
`x_raw=node_50.tpdo[2][0x3010.1].raw`

sam-tj, 2023-09-13:
Hi,
I have the same question and problem. When i use the parameter name I get the following error. 

```

imu_reader_canopen.py 37
Traceback (most recent call last):
  File "/home/tester/test_ws/pycan/imu_reader/imu_reader_canopen.py", line 37, in <module>
    node.tpdo.save()
  File "/home/tester/.local/lib/python3.10/site-packages/canopen/pdo/base.py", line 61, in save
    pdo_map.save()
  File "/home/tester/.local/lib/python3.10/site-packages/canopen/pdo/base.py", line 401, in save
    self.map_array[subindex].raw = (var.index << 16 |
  File "/home/tester/.local/lib/python3.10/site-packages/canopen/variable.py", line 89, in raw
    self.data = self.od.encode_raw(value)
  File "/home/tester/.local/lib/python3.10/site-packages/canopen/variable.py", line 41, in data
    self.set_data(data)
  File "/home/tester/.local/lib/python3.10/site-packages/canopen/sdo/base.py", line 132, in set_data
    self.sdo_node.download(self.od.index, self.od.subindex, data, force_segment)
  File "/home/tester/.local/lib/python3.10/site-packages/canopen/sdo/client.py", line 161, in download
    fp.close()
  File "/home/tester/.local/lib/python3.10/site-packages/canopen/sdo/client.py", line 389, in write
    response = self.sdo_client.request_response(request)
  File "/home/tester/.local/lib/python3.10/site-packages/canopen/sdo/client.py", line 85, in request_response
    return self.read_response()
  File "/home/tester/.local/lib/python3.10/site-packages/canopen/sdo/client.py", line 73, in read_response
    raise SdoAbortedError(abort_code)
canopen.sdo.exceptions.SdoAbortedError: Code 0x06040041, Object cannot be mapped to the PDO
going to exit... stopping...
``` 

Here is my Record details. 

```
[6001]
ParameterName=gyroscope
ObjectType=0x9
;StorageLocation=PERSIST_COMM
SubNumber=0x4

[6001sub0]
ParameterName=Highest sub-index supported
ObjectType=0x7
;StorageLocation=RAM
DataType=0x0005
AccessType=ro
DefaultValue=0x03
PDOMapping=0

[6001sub1]
ParameterName=gyro_x
ObjectType=0x7
;StorageLocation=RAM
DataType=0x0003
AccessType=ro
DefaultValue=0
PDOMapping=1

[6001sub2]
ParameterName=gyro_y
ObjectType=0x7
;StorageLocation=RAM
DataType=0x0003
AccessType=ro
DefaultValue=0
PDOMapping=1

[6001sub3]
ParameterName=gyro_z
ObjectType=0x7
;StorageLocation=RAM
DataType=0x0003
AccessType=ro
DefaultValue=0
PDOMapping=1 
```

It would be really helpful if you can provide a small example to get ParameterName from the OD directly and then read it in a loop.

Thanks in advance.