canopen-python/canopen #400: RPDO not waiting for SYNC before triggering

jitaxis, 2023-10-17:
I have struggled to get two motors to run synchronously over the past few days.
The code below is me messing around trying to get it to work, so I apologise for how unorganised it is. If anyone is able to take a look and tell me what I am doing wrong it would be greatly appreciated.

`#!/usr/bin/env python3
import canopen
import time

OD_CONFIG_INDEX             = 0x2010
OD_AUTO_TUNING_SUBINDEX     = 0x10
OD_DIN_INDEX_2              = 0x2020
OD_DIN_MODE0_SUBINDEX       = 0x0D
OD_DIN_MODE1_SUBINDEX       = 0x0E
OD_MOTOR_INFO_INDEX         = 0x2010
OD_MOTOR_INFO_TEMP_SUBINDEX = 0x19
CONTROLL_WORD_FOR_IP        = 0x1F

# Create a CANopen network object
network = canopen.Network()
# Connect to the virtual CAN network interface (adjust the interface based on your setup)
network.connect(bustype='socketcan', channel='can0',bitrate=500000)
node_2 = canopen.BaseNode402(2, '/KINCO-JD-FD.eds')
node_3 = canopen.BaseNode402(3, '//KINCO-JD-FD.eds')
# Add a node_2 to the network
network.add_node(node_2)  # Replace 'eds_file.eds' with your EDS file
network.add_node(node_3)  # Replace 'eds_file.eds' with your EDS file

# Reset network
node_2.nmt.state = 'RESET COMMUNICATION'
print('node_2 state 0) = {0}'.format(node_3.nmt.state))
node_3.nmt.state = 'RESET COMMUNICATION'
print('node_3 state 0) = {0}'.format(node_3.nmt.state))

timeout = time.time() + 10
node_2.nmt.state = 'PRE-OPERATIONAL'
while node_2.nmt.state != 'PRE-OPERATIONAL':
    if time.time() > timeout:
        raise Exception('Timeout when trying to change state')
    time.sleep(0.1)
print('node_2 state 1) = {0}'.format(node_2.nmt.state))

timeout = time.time() + 10
node_3.nmt.state = 'PRE-OPERATIONAL'
while node_3.nmt.state != 'PRE-OPERATIONAL':
    if time.time() > timeout:
        raise Exception('Timeout when trying to change state')
    time.sleep(0.1)
print('node_3 state 1) = {0}'.format(node_3.nmt.state))

node_2.load_configuration()
node_3.load_configuration()

print('node_2 state 3) = {0}'.format(node_2.nmt.state))
print('node_3 state 3) = {0}'.format(node_3.nmt.state))

node_2.setup_402_state_machine()
node_3.setup_402_state_machine()
# device_name = node_3.sdo[0x1008].raw
# vendor_id = node_3.sdo[0x1018][1].raw
# print(device_name)
# print(vendor_id)

node_2.tpdo[1].clear()
node_2.tpdo[1].enabled = True
node_2.tpdo[1].add_variable('Statusword')
node_2.tpdo[1].add_variable(0x6040)
node_2.tpdo[1].add_variable(0x6063)
node_2.tpdo[1].cob_id  = 0x182
# node_2.tpdo[1].add_variable('Some group', 'Some subindex')
node_2.tpdo[1].trans_type = 0x05
# node_2.tpdo[1].event_timer = 10
# Save new PDO configuration to node_2
node_2.tpdo[1].save()
print ("TX_ID: ",hex(node_2.sdo[0x1800][0x01].raw))
print ("TX type: ",hex(node_2.sdo[0x1A00][0x02].raw))
print ("Tpdo mapping 1: ",hex(node_2.sdo[0x1A00][0x01].raw))

node_2.rpdo[1].clear()
node_2.rpdo[1].enabled = True
node_2.rpdo[1].add_variable('Interpolation_data_record','Interpolation_data_record_x1 1')
node_2.rpdo[1].trans_type = 0x05
# node_2.rpdo[1].inhibit_time = 10
node_2.rpdo[1].cob_id  = 0x202
node_2.rpdo[1].save()
print ("RX_ID: ",hex(node_2.sdo[0x1400][0x01].raw))
print ("TX type: ",hex(node_2.sdo[0x1400][0x02].raw))
print ("rpdo mapping 1: ",hex(node_2.sdo[0x1600][0x01].raw))
# node_2.sdo[0x1600][0x01].raw = 0x60C10120   #INDEX SUB-index NO. bits
# node_2.sdo[0x1400][0x02].raw = 0x01         #Transmit on every sync
# node_2.sdo[0x1400][0x03].raw = 10           #ignore anything faster than x*100us e.g 10 = 1ms

node_3.rpdo[1].clear()
node_3.rpdo[1].add_variable('Interpolation_data_record','Interpolation_data_record_x1 1')
node_3.rpdo[1].trans_type = 0x01
node_3.rpdo[1].enabled = True
node_3.rpdo[1].save()

timeout = time.time() + 10
node_2.nmt.state = 'OPERATIONAL'
while node_2.nmt.state != 'OPERATIONAL':
    if time.time() > timeout:
        raise Exception('Timeout when trying to change state')
    time.sleep(0.1)
print('node_2 state 4) = {0}'.format(node_2.nmt.state))

timeout = time.time() + 10
node_3.nmt.state = 'OPERATIONAL'
while node_3.nmt.state != 'OPERATIONAL':
    if time.time() > timeout:
        raise Exception('Timeout when trying to change state')
    time.sleep(0.1)
print('node_3 state 4) = {0}'.format(node_3.nmt.state))

# -----------------------------------------------------------------------------------------
# node_2.nmt.start_node_guarding(0.01)
# node_3.nmt.start_node_guarding(0.01)

current_mode = 0x0007
    ##### NODE 2 SETUP #####
    #  Set Control Word
control_word = int.from_bytes(node_2.sdo.upload(0x2020, 0x0F), 'big')
print(f'Control word: {hex(control_word)}')
control_word = 0x000F
node_2.sdo.download(0x2020, 0x0F, control_word.to_bytes(1, 'big'))
print(f'Control word: {hex(control_word)}')

din_mode0 = current_mode
din_mode1 = current_mode
print (din_mode0.to_bytes(1, 'big'))
node_2.sdo.download(OD_DIN_INDEX_2, OD_DIN_MODE0_SUBINDEX, din_mode0.to_bytes(1, 'big'))
node_2.sdo.download(OD_DIN_INDEX_2, OD_DIN_MODE1_SUBINDEX, din_mode1.to_bytes(1, 'big'))

node_2.sdo[0x60C2][0x01].raw = 100
print(f'IP Interpolation time: {hex(node_2.sdo[0x60C2][0x01].raw)}')
    # -4 = mirco-seconds, -3 = milli-seconds ...
node_2.sdo[0x60C2][0x02].raw = -3
print(f'IP Power of ten of the interpolation time: {hex(node_2.sdo[0x60C2][0x02].raw)}')
# print(f'IP interval: {hex(node_2.sdo[0x60C4][0x06].raw)}')

node_2.sdo[0x2010][0x2].raw = 0x02
    #  Simulation of digital input 8bit register each bit repersents a input to set
node_2.sdo[0x2010][0x2].raw = 0x01

control_word = 0x001F
# control_word = 0x000F
node_2.sdo.download(0x2020, 0x0F, control_word.to_bytes(1, 'big'))

    ##### NODE 3 SETUP #####

    #  Set Control Word
control_word = int.from_bytes(node_3.sdo.upload(0x2020, 0x0F), 'big')
print(f'Control word: {hex(control_word)}')
control_word = 0x000F
node_3.sdo.download(0x2020, 0x0F, control_word.to_bytes(1, 'big'))
print(f'Control word: {hex(control_word)}')

din_mode0 = current_mode
din_mode1 = current_mode
print (din_mode0.to_bytes(1, 'big'))
node_3.sdo.download(OD_DIN_INDEX_2, OD_DIN_MODE0_SUBINDEX, din_mode0.to_bytes(1, 'big'))
node_3.sdo.download(OD_DIN_INDEX_2, OD_DIN_MODE1_SUBINDEX, din_mode1.to_bytes(1, 'big'))

node_3.sdo[0x60C2][0x01].raw = 100
print(f'IP Interpolation time: {hex(node_3.sdo[0x60C2][0x01].raw)}')
    # -4 = mirco-seconds, -3 = milli-seconds ...
node_3.sdo[0x60C2][0x02].raw = -3
print(f'IP Power of ten of the interpolation time: {hex(node_3.sdo[0x60C2][0x02].raw)}')
# print(f'IP interval: {hex(node_3.sdo[0x60C4][0x06].raw)}')

node_3.sdo[0x2010][0x2].raw = 0x02
    #  Simulation of digital input 8bit register each bit represents a input to set
node_3.sdo[0x2010][0x2].raw = 0x01
# node_3.sdo[0x2010][0x2].raw = 0x00
control_word = 0x001F
# control_word = 0x000F
node_3.sdo.download(0x2020, 0x0F, control_word.to_bytes(1, 'big'))

# Transmit SYNC every 100 ms
network.sync.start(0.5)
position_2 = node_2.sdo[0x6063].raw
position_3 = node_3.sdo[0x6063].raw
print (position_2,position_3)

while True:
    try:
        network.check()
    except Exception:
        break
    node_2.tpdo[1].wait_for_reception()
    pos = node_2.tpdo[1][0x6063].raw
    print(f'tpdo position: {pos}')
    # node_2.sdo[0x607A].raw = position
    node_2.rpdo[1]['Interpolation_data_record.Interpolation_data_record_x1 1'].phys = position_2
    print (position_2,node_2.rpdo[1].data)
    node_2.rpdo[1].transmit()
    # print(f'target_vel 2: {node_2.sdo["Target_position"].raw}')
    node_3.rpdo[1]['Interpolation_data_record.Interpolation_data_record_x1 1'].raw = position_3
    node_3.rpdo[1].transmit()
    # print(f'target_vel 3: {node_3.sdo["Target_position"].raw}')
    position_2 = position_2 + 1000
    position_3 = position_3 + 1000
    time.sleep(0.1)

# Disconnect from CAN bus
print('going to exit... stopping...')
if network:

    for node_2_id in network:
        node_2 = network[node_2_id]
        node_2.nmt.state = 'PRE-OPERATIONAL'
        node_2.nmt.stop_node_2_guarding()
    network.sync.stop()
    network.disconnect()`
    
    The .EDS file if required:

[KINCO-JD-FD.zip](https://github.com/christiansandberg/canopen/files/12965797/KINCO-JD-FD.zip)

Also when setting the position values, I have seen .raw and .phys used, what is the difference?

jitaxis, 2023-10-26:
I cut the code down a bit to just toggle an output on the motor driver. The output still changes on the RPDO send and not the SYNC. The transmission type object is 1 when polled. 
I am really struggling in getting this sorted. If anyone is able to show me how they set up SYNC'd RPDO communication that would be amazing. 
Thanks.

```
#!/usr/bin/env python3
import canopen
import time

# Create a CANopen network object
network = canopen.Network()

# Connect to the virtual CAN network interface (adjust the interface based on your setup)
network.connect(bustype='socketcan', channel='can0',bitrate=500000)
node_2 = canopen.BaseNode402(2, '/home/jtobias/ros2_ws/src/harvester/bhm_can_interface/config/single-kinco-fd124s/KINCO-JD-FD.eds')
# Add a node_2 to the network
network.add_node(node_2)  # Replace 'eds_file.eds' with your EDS file
# Reset network
node_2.nmt.state = 'RESET COMMUNICATION'
print('node_2 state 0) = {0}'.format(node_2.nmt.state))

timeout = time.time() + 10
node_2.nmt.state = 'PRE-OPERATIONAL'
while node_2.nmt.state != 'PRE-OPERATIONAL':
    if time.time() > timeout:
        raise Exception('Timeout when trying to change state')
    time.sleep(0.1)
print('node_2 state 1) = {0}'.format(node_2.nmt.state))
node_2.load_configuration()
node_2.setup_402_state_machine()

# node_2.sdo[0x3011][0x01].raw = 0x01
# node_2.sdo[0x3011][0x02].raw = 0x01

# time.sleep(5)
node_2.tpdo[1].clear()
node_2.tpdo[1].add_variable('Statusword')
node_2.tpdo[1].add_variable(0x6040)
node_2.tpdo[1].add_variable(0x6063)
# node_2.tpdo[1].cob_id  = 0x182
node_2.tpdo[1].trans_type = 2   # 0 stops the transmission
node_2.tpdo[1].enabled = True
node_2.tpdo[1].save()
print ("TX_ID: ",hex(node_2.sdo[0x1800][0x01].raw))
print ("TX type: ",hex(node_2.sdo[0x1800][0x02].raw))
print ("Tpdo mapping 1: ",hex(node_2.sdo[0x1A00][0x01].raw))
time.sleep(2)
print (f"Node 2 Sync ID: {hex(node_2.sdo[0x1400][0x01].raw)}")
node_2.sdo[0x1400][0x01].raw = 0x7FF
node_2.sdo[0x1600][0x00].raw = 0x00
node_2.sdo[0x1400][0x02].raw = 1
node_2.sdo[0x1600][0x01].raw = 0x20100E10
node_2.sdo[0x1600][0x00].raw = 0x01
node_2.sdo[0x1400][0x01].raw = 0x202
time.sleep(1)
print (f"Node 2 Tx type: {hex(node_2.sdo[0x1400][0x02].raw)}")
print (f"Node 2 Number Of entries: {hex(node_2.sdo[0x1600][0x00].raw)}")
print (f"Node 2 ID: {hex(node_2.sdo[0x1400][0x01].raw)}")

    # This is used to set the controlword
node_2.sdo[0x2010][0x2].raw = 0x01
time.sleep(1)
print('node_2 = {0}'.format(node_2.state))
error_code = 0x00

timeout = time.time() + 10
node_2.nmt.state = 'OPERATIONAL'
while node_2.nmt.state != 'OPERATIONAL':
    if time.time() > timeout:
        raise Exception('Timeout when trying to change state')
    time.sleep(0.1)
print('node_2a = {0}'.format(node_2.nmt.state))

timeout = time.time() + 10
# -----------------------------------------------------------------------------------------
dout_val = 0x0001
while True:
    try:
        network.check()
    except Exception:
        break
    # Read the state of the Statusword
    print('node_2 state) = {0}'.format(node_2.nmt.state))
    print(f'statusword: {hex(node_2.sdo[0x6041].raw)}')
    print(f'controlword: {hex(node_2.sdo[0x6040].raw)}')
    print(f'Error Codes: {hex(node_2.sdo[0x2601].raw)}')
    print(f'ECAN sync: {hex(node_2.sdo[0x3011][0x01].raw)}')
    print(f'ECAN clock: {hex(node_2.sdo[0x3011][0x02].raw)}')

    dout_val ^= 0x0001
    print (dout_val)
    node_2.pdo.network.send_message(0x202, dout_val.to_bytes(2, 'little'))
    print(f'Douts 2,3: {hex(node_2.sdo[0x2010][0x14].raw)}')
    print(f'tx type: {hex(node_2.sdo[0x1400][0x02].raw)}')
    time.sleep(2)
    print(f'Douts 2,3: {hex(node_2.sdo[0x2010][0x14].raw)}')
    print ("Sync")
    network.sync.transmit()
    print(f'Douts 2,3: {hex(node_2.sdo[0x2010][0x14].raw)}')
    time.sleep(1)
    print ("Finished wait")
    time.sleep(2)

# Disconnect from CAN bus
print('going to exit... stopping...')
if network:

    for node_2_id in network:
        node_2 = network[node_2_id]
        node_2.nmt.state = 'PRE-OPERATIONAL'
        # node_2.nmt.stop_node_2_guarding()
    network.sync.stop()
    network.disconnect()
    ```

theveloped, 2023-11-23:
Hi @jitaxis,

Curious to here if you have been able to get synced messages to be correctly sent to your drive? I'm looking to do the same but am running into similar issues were the position setpoints are not correctly sent at sync in my case to a ASDA-A2 servo drive.

jitaxis, 2023-11-29:
I got it working. I started using more manual commands for sending CAN messages.
I first set up the RPDOs on the drivers manually through the driver software, but you could do this using PDOs during the Operational stage for the driver.

The steps I took to get synced drivers are as follows:

1. Reset Drivers (Using NMT commands to STOP,sleep(0.5), set to Preoperational, load_config)
2. Setup any PDOs
3. Start drivers (Using NMT set into Operational mode)
4. In a while loop I manually send a SYNC msg, and after it has been sent (i wait a very short period) I send my RPDOs

closed 2023-11-29 by jitaxis