canopen-python/canopen #403: SdoClient.download: allow block transfer mode

dentoyan, 2023-11-02:
Allow large file  transfer when using download. Block transfer mode is correctly implemented.
The block_transfer argument must be set to True for the open method. This  will speed up the transfer significantly.

sveinse, 2024-05-19:
Hi! What is the question here? Isn't the `block_transfer` mode of `open()` working on downloads? Neither `upload()` nor `download()` of `SdoClient` use block transfer mode, you have to call `open()` manually.

dentoyan, 2024-05-23:
Please review my commit  https://github.com/christiansandberg/canopen/pull/419/commits/07d82a66a6c752a29cb002b5a1bf89ed17904bf9

acolomb, 2024-05-23:
Sorry, I don't get what the intention is here, either.  You want `SdoClient.download()` to default to block transfer mode?  That doesn't make sense in the vast majority of use cases.  And for those where it does, we have a solution by using `open()` directly from your application code.

sveinse, 2024-05-23:
What PR is this from?

Can I maybe suggest to add `block_transfer: bool = False` as an additional argument to `SdoClient.upload()` and `SdoClient.download()`.? If you do , please remember that the prototypes in `SdoBase.upload()` and `SdoBase.download()` must be updated too.

dentoyan, 2024-05-29:
Adding an additional argument is a good idea.
Block Transfer: multiple segments are confirmed by only 1 confirm message.
On Initiate Block Download command the server returns the number of segments per block. So the client can select the optimal transfer mode.