canopen-python/canopen #410: test_abort

OliveiraICTS, 2023-12-27:
Hello,

I´m studing CANopen protocols and I found your excellent Source code in Python on github.
Now I´m studying SDO session and matching with your code and everyghing looks okay, except one.
on test_sdo.py the function test_abort you are sending a the same message to the same endpoint index 1018 subindex 01 used on test expedicted_upload. It is included on sample.eds file used as reference.

Question: You are raising the exception code 0x06090011 just for test?

The reason is that 0x1018 0x01 is a mandatory and the above exception code is rased if its subindex not found on dictionary.  Is it right?

sveinse, 2024-05-19:
With the disclaimer that I didn't write the test mentioned: if an object with a subindex not found it will send the abort code 0x06090011, which the test does, yes. 0x1018:01 is mandatory, but it doesn't make the test invalid.

acolomb, 2024-05-23:
I don't see an issue here, please comment if anything needs adjustment, then we can re-open it.

closed 2024-05-23 by acolomb