canopen-python/canopen #425: Minimum python version is 3.10

sveinse, 2024-04-26:
The source contains type annotations using the `|` operator, which was introduced in python 3.10. Effectively this means that the minimum python version for canopen is `>=3.10` and not 3.6 as specified in `setup.cfg`. If this is intentional, then I propose we change the setup information accordingly.

I have verified that it runs on `3.11` and `3.12` by running tox with the included pytest test suite.

acolomb, 2024-04-26:
That is unfortunate, as it is really unnecessary to bump the Python version requirement just because of this usage. And as far as I found, there is actually only one instance of it, here:
https://github.com/christiansandberg/canopen/blob/dc8a4ca79a45db4c5f56fe0db8f3634202b63ada/canopen/network.py#L36

Did you see any other occurrences?

I'd prefer to fix that instead. As much as I like the improved type annotation features, as a library we should stay conservative and adopt the new standard only when it brings enough benefit or is used more consistently throughout the code.

sveinse, 2024-04-26:
I have checked that with the patch of `canopen/network.py`

```
-    def __init__(self, bus: can.BusABC | None = None):
+    def __init__(self, bus: Optional[can.BusABC] = None):
```

It will work fine all versions down to python 3.6.

acolomb, 2024-04-28:
Thanks for checking.  The change was done in https://github.com/christiansandberg/canopen/commit/b55ce5cac8d645ecb14b622323ae6136e561966c.

closed 2024-04-28 by acolomb