canopen-python/canopen #436: Inconsistent data truncation of unsupported data types in SDO upload

sveinse, 2024-05-18:
I'm having problems reading SDO objects that are using UNSIGNED48. It used to work before PR #395 by @ljungholm - but it turns out to be coincidental.

https://github.com/christiansandberg/canopen/blob/24df6e846a4e71508fd82c378d1385d65f000c94/canopen/sdo/client.py#L123-L133

I believe the root cause is that canopen doesn't support the UNSIGNED48 type (which I'll contribute in another PR). The unfortunate behavior is that due to the unknown type, the data gets truncated to 1 byte. It didn't do that before #395 .

```
>> upload((8192, 2), {})
<-- 602     :  40 00 20 02 00 00 00 00
--> 582     :  41 00 20 02 08 00 00 00
  << read_response() = b'A\x00 \x02\x08\x00\x00\x00'
<-- 602     :  60 00 00 00 00 00 00 00
--> 582     :  00 b2 01 20 02 91 12 00
  << read_response() = b'\x00\xb2\x01 \x02\x91\x12\x00'
<-- 602     :  70 00 00 00 00 00 00 00
--> 582     :  1d 00 00 00 00 00 00 00
  << read_response() = b'\x1d\x00\x00\x00\x00\x00\x00\x00'
     data=b'\xb2\x01 \x02\x91\x12\x00\x00'
     response_size = 8
     var.datatype = 25   # UNSIGNED48 (unsupported)
     var_size = 1   # Because len(var) on unknown datatypes is 8
     data = b'\xb2'   # Truncated due to ^
<< upload() = b'\xb2'
```

Observations:
* Reading SDO objects _without_ OD record will preserve data as-is
* Reading SDO objects _with_ OD, but of unsupported type will return data of size 1
* `ODVariable.__len__()` return 8 on unknown types, while it reports number of bits in the other types. This is why it gets truncated to one byte.

Would you agree that `upload()` should return the the full data for both of the first use cases? Is the fix that `ODVariable.__len__()` on unknown datatypes should return 64?

sveinse, 2024-05-20:
> Would you agree that `upload()` should return the the full data for both of the first use cases? Is the fix that `ODVariable.__len__()` on unknown datatypes should return 64?

PR #440 implements four test cases (`test_unknown_*()`) that address this issue. The test fails in the PR because it contains longer data. IMHO canopen shouldn't truncate any messages it encounters for unknown datatypes, it should pass them through. Especially when it _is_ passed if there is no OD entry for it.

Why do we have the logic of `if var.data_type not in objectdictionary.DATA_TYPES:` test and then proceed with truncating the message?

christiansandberg, 2024-05-20:
Maybe use 64 bits instead of 8 bytes is the easiest solution. Hopefully it shouldn't happen too often after 48 bits support is added.

sveinse, 2024-05-20:
It depends on not getting a future datatype which is larger than 64 bits. If we chose to cap it at 64, I would suggest that we warn about it to the logger that the data has been capped with an unknown datatype. (And the test in PR #440 must be altered to make it pass.)

acolomb, 2024-05-23:
I'm not really fond of replacing the fall-back 8 bits value by 64.  It makes sense for any type to have at least 8 bits, since one octet is the smallest unit supported by SDO (disregarding PDO mapping with finer granularity here).  Using any other fall-back value is arbitrary and will break unexpectedly when it does.

We should rather fix the min(OD, SDO) length assumption to only get applied when the OD size is actually meaningful and not itself a fall-back value.

christiansandberg, 2024-05-23:
I agree, better to check if size is known or not.

sveinse, 2024-05-23:
I attempted
```python
        var = self.od.get_variable(index, subindex)
        if var is not None:
            if var.data_type not in objectdictionary.DATA_TYPES:
                # Use max to determine enough space for the unknown data type
                var_size = max(len(var) // 8, response_size)
                if response_size is None or var_size < response_size:
                    # This wil never happen...
                    data = data[0:var_size]
        return data
```

But it turned out a bit non-sensical. So I'm kind of back at: Why do we even truncate? Why do we want to reduce the data for something we don't know what is?

acolomb, 2024-05-23:
Just a quick note, the truncating was introduced to avoid an exception from the struct methods. They are quite strict about the buffer length, but could be tricked otherwise by slicing.

Will need to take another good look at the alternatives and code flow...

sveinse, 2024-05-23:
I would assume that _known_ data types can or should be truncated in case of excessive data. Here we're talking about unknown data types with no reference to validate its proper size.

acolomb, 2024-05-27:
I'm really still uncertain what the actual problem here is.  `SdoClient.upload()` is actually independent from the `ODVariable`.  It used to (before #395) only consult the OD when the uploaded data was of unspecified size.  #395 changed that to apply even when an upload size was indicated, which I think is not always correct.  For broken devices that send e.g. an UNSIGNED8 as UNSIGNED32 with four data bytes, it does make sense to truncate the data instead of failing during `unpack()`.  But for any type not based on `struct.Struct`, truncating is usually harmful.

The real culprit here is IMHO the check `var.data_type not in objectdictionary.DATA_TYPES`.  That applies for any unknown OD entry type as well, but should actually be limited to types where the length is known in advance.  So better to check for `var.data_type in var.STRUCT_TYPES` here, assuming that all known-length data types will be unpacked using structs.  Even better to encapsulate this into a new property `ODVariable.length_known`.

And maybe this whole check should be moved to `SdoVariable.get_data()`, where we are certain that an associated `ODVariable` is actually available?  For the "OD-independent" (as stated by the docstring) `.upload()` method, consulting the OD is a bit inconsistent.

acolomb, 2024-05-27:
I think the problem is to more clearly define what is a "known" data type.  The check against `objectdictionary.DATA_TYPES` actually just matches `(VISIBLE_STRING, OCTET_STRING, UNICODE_STRING, DOMAIN)`.  Those are the defined ones which are not parsed by a struct, and the check also matches undefined data type entries.  We should invert that and apply the truncation only for known data types where a struct gets used for parsing, I think.

Also, your second case is impossible, as the data type is defined in the OD entry, thus without one we cannot know.

sveinse, 2024-05-27:
> I think the problem is to more clearly define what is a "known" data type. The check against `objectdictionary.DATA_TYPES` actually just matches `(VISIBLE_STRING, OCTET_STRING, UNICODE_STRING, [...]

Yes, I think that is a good idea. We must except array-types from this truncation, being `(VISIBLE_STRING, OCTET_STRING, UNICODE_STRING, DOMAIN)`.

> Also, your second case is impossible, as the data type is defined in the OD entry, thus without one we cannot know.

Yes, you're right. Without OD one cannot know the data type. I missed the fact. It is possible thou to read an object with SDO that is missing from the OD and thus its data type is missing. Its data should definitely not be truncated.

sveinse, 2024-05-27:
It must look up the OD to get the data type, so in order for `SdoClient.upload()` to do _any_ length validation, it must use the OD.

IMHO `SdoClient.upload()` is just a raw producer of `bytes` using an SDO upload request. The deserialization step into native python objects happens later. So I think I lean towards not doing any validation in `SdoClient.upload()`.

Today the deserialization happens in `ODVariable.decode_raw()` with its call from the `Variable.raw` property. This is where the stream from the SDO upload should be validated, I think.

acolomb, 2024-06-03:
The only reason for doing validation in `.upload()` is that we know the indicated data set size there.  The check prior to #395 actually only kicked in when that size was `None`, i.e. the server didn't specify how many data bytes should be interpreted.

When moving the validation to `decode_raw()` completely, we lose that bit of information.  But frankly, I'm not sure we really need it at all.  Trying to summarize it into a table again (may need to scroll horizontally):

No. | Type in OD | SDO Size Indication<br>(`response_size`) | SDO Payload<br>(`len(data)`) | Action in<br>`decode_raw()` | Use Case
-- | -- | -- | -- | -- | --
1 | `in var.STRUCT_TYPES` | unspecified | `== len(var)` | OK, unpack | normal
2 | `in var.STRUCT_TYPES` | unspecified | `< len(var)` | pad zeros right, unpack | partial (compressed?) response
3 | `in var.STRUCT_TYPES` | unspecified | `> len(var)` | truncate right, unpack | normal, depends on OD
4 | `in var.STRUCT_TYPES` | `== len(var)` | (don’t care) | OK, unpack | normal
5 | `in var.STRUCT_TYPES` | `< len(var)` | (don’t care) | pad zeros right, unpack | partial (compressed?) response
6 | `in var.STRUCT_TYPES` | `> len(var)` | (don’t care) | truncate right, unpack | quirk, always 4-byte for small vars
7 | `in object_dictionary.DATA_TYPES` | unspecified | use as length | `bytes(data)` | normal, streaming until complete
8 | `in object_dictionary.DATA_TYPES` | specified | (don’t care) | `bytes(data[:response_size])` | normal, discarding extra (unused) bytes
9 | neither STRUCT nor DATA | unspecified | use as length | `bytes(data)` | unknown data type
10 | neither STRUCT nor DATA | specified | (don’t care) | `bytes(data[:response_size])` | unknown data type, discarding extra bytes
11 | no OD entry | unspecified | use as length | `bytes(data)` | unknown data type
12 | no OD entry | specified | (don’t care) | `bytes(data[:response_size])` | unknown data type, discarding extra bytes

Notes:
* (1) to (3) and (4) to (6) are equivalent here, which is why we don't really need to distinguish the `response_size`  being specified while decoding.
* (2) and (5) are mostly useful in PDO context, where a numeric variable can be mapped partially (e.g. UNSIGNED32 object with small values mapping only 16 bits to PDO).  We might want to differentiate this between PDO and SDO then?
* (6) is essentially what #395 tried to fix.  Again, by handling the quirky responses in this way we make it equivalent to (3).
* (7) to (12) all have in common that they are `not in var.STRUCT_TYPES` (also if `var is None`), thus this is the only check we would really need.  `not in objectdictionary.DATA_TYPES` (as currently coded) however doesn't catch all cases, especially unknown bit widths, that's why new tests fail without #440.

Does that all seem correct?  Would it be okay for you if we used this table as basis for the implementation and tested each case?

sveinse, 2024-06-04:
This is a really good overview. There are two independent usages of length involved here

* **Declared SDO upload size (`response_size`) vs. actual data (`len(data)`).** If these don't match, I am inclined to consider this a protocol violation. (I speculate if the SDO abort code `0x05040002` is intended for such size mismatches in block mode).

* **Actual data size (`len(data)`) vs. expected OD size (`len(var)`)**. If there is too little data to decode the format, then, well, there is not enough information and this should be an exception. If there is excessive data, it can be truncated. But only if the data type is known. For anything else, the data should be passed transparently through.

acolomb, 2024-06-10:
I'd really like to get this one fixed soon, but right now we only have ideas and not a complete and well-tested implementation draft of these last suggestions.  Therefore I think we should focus on getting a release out the door to avoid problems like #455 and come back to this afterwards.

OK for you @sveinse?

sveinse, 2024-06-11:
@acolomb Yes, let's not delay the release because of this. Current behavior is what it is and we're not introducing any regressions by not fixing this issue.

sveinse, 2024-06-16:
@friederschueler not sure I'd label this one as an enhancement. Its describing unexpected behavior, so it's a bug.

sveinse, 2025-06-14:
I think we should pick up this one soon. It was a great effort earlier to get all the missing datatypes implemented.

I'm having a system with manufacturer specific datatypes. Due to how the current logic is implemented it gets truncated. A working example against HW:

```py
        od = ObjectDictionary()
        record = ODRecord("Record", 0x2020)
        var = ODVariable("Variable", 0x2020, 1)
        var.data_type = 0x40  # Custom manufacturer data-type. Returns 12 bytes
        record.add_member(var)  # <-- If this is commented out, the code returns right value
        od.add_object(record)
        node = network.add_node(2, od)
        data = await node.sdo.aupload(0x2020, 1)
        # data == b'2' (one byte), which is due to the truncation into 8 bits
```

> Trying to summarize it into a table again (may need to scroll horizontally):

What's missing is handling of use case (9) - (12), so I'll focus on them.

* (9) Truncates to 1 byte
* (10) Truncates to 1 byte (the above example is in this class)
* (11) Works fine, returns 12 bytes
* (12) Works fine, returns 12 bytes

It is fully valid for users to define custom data types and we should not interfere with the data in those instances.

sveinse, 2025-06-15:
> What you're fixing in [#447](https://github.com/canopen-python/canopen/pull/447) touches several of these cases. The changed test is actually about (6), totally unrelated to the case of a missing OD entry or unknown data type.

> So I think we should in fact concentrate on the wrongfully truncating cases, (9) and (10).

The changed test `test_size_not_specified` where to make the test case pass, as #447 moves the expected data truncation to `ODVariable.decode_raw()`. However, tests `test_unknown_datatype32` and `test_unknown_datatype112` were reenabled, which tests use case (10) and (12). It should be failry easy to add for (9) and (11).

> ...not before the next release.

Well... We said the same thing a release ago. 😜