canopen-python/canopen #437: Object Dictionary module | UNSIGNED48 | CANOpen Lift 417 IO-Port EDS

joe0799, 2024-05-19:
Please add UNSIGNED48

File : ~/.local/lib/python3.11/site-packages/canopen/objectdictionary/__init__.py

**TOP Part**

```
UNSIGNED48 = 0x19  # Assuming 0x19 is the identifier for UNSIGNED48

def unpack_unsigned48(data):
    """Unpack 48-bit unsigned integer from bytes."""
    assert len(data) == 6, "UNSIGNED48 data should be exactly 6 bytes"
    return int.from_bytes(data, byteorder='little', signed=False)

def pack_unsigned48(value):
    """Pack 48-bit unsigned integer into bytes."""
    assert 0 <= value < (1 << 48), "Value out of range for UNSIGNED48"
    return value.to_bytes(6, byteorder='little')
```

**Decode-Part:**
 ```
   def decode_raw(self, data: bytes) -> Union[int, float, str, bytes, bytearray]:
        if self.data_type == UNSIGNED48:
            return unpack_unsigned48(data)
        elif self.data_type == VISIBLE_STRING:
            return data.rstrip(b"\x00").decode("ascii", errors="ignore")
        elif self.data_type == UNICODE_STRING:
            # Is this correct?
            return data.rstrip(b"\x00").decode("utf_16_le", errors="ignore")
        elif self.data_type in self.STRUCT_TYPES:
            try:
                value, = self.STRUCT_TYPES[self.data_type].unpack(data)
                return value
            except struct.error:
                raise ObjectDictionaryError(
                    "Mismatch between expected and actual data size")
        else:
            # Just return the data as is
            return data
```
`
**Encode Part**
 ```
   def encode_raw(self, value: Union[int, float, str, bytes, bytearray]) -> bytes:
        if isinstance(value, (bytes, bytearray)):
            return value

        elif self.data_type == UNSIGNED48:
            return pack_unsigned48(value)

        elif self.data_type == VISIBLE_STRING:
            return value.encode("ascii")
        elif self.data_type == UNICODE_STRING:
            # Is this correct?
            return value.encode("utf_16_le")
        elif self.data_type in self.STRUCT_TYPES:
            if self.data_type in INTEGER_TYPES:
                value = int(value)
            if self.data_type in NUMBER_TYPES:
                if self.min is not None and value < self.min:
                    logger.warning(
                        "Value %d is less than min value %d", value, self.min)
                if self.max is not None and value > self.max:
                    logger.warning(
                        "Value %d is greater than max value %d",
                        value,
                        self.max)
            try:
                return self.STRUCT_TYPES[self.data_type].pack(value)
            except struct.error:
                raise ValueError("Value does not fit in specified type")
        elif self.data_type is None:
            raise ObjectDictionaryError("Data type has not been specified")
        else:
            raise TypeError(
                "Do not know how to encode %r to data type %Xh" % (
                    value, self.data_type))

```
`

sveinse, 2024-05-19:
Related to #436 - and I am currently working on adding the missing datatypes to canopen

joe0799, 2024-05-19:
> Related to #436 - and I am currently working on adding the missing datatypes to canopen

UNSIGNED 48 is integrated with the code snippet

PR #440 (2024-06-11): Implement the remaining canopen datatypes
* Add datatype defs for the canopen standard types
* Rename datatypes_24bit.py to datatypes_struct.py
* Replace Unsigned24 and Interger24 by generic UnsignedN and IntegerN respectively
* Add EDS-file containing all datatypes
* Added tests for encoding and decoding all datatypes
* Added tests for SDO uploads of all datatypes

Related to #436 and #439. This issue will close #437. Closes #320
closed 2024-06-11 by acolomb