canopen-python/canopen #438: Type annotation preference

sveinse, 2024-05-19:
Hi. I'm looking for some advice on what style of type annotation that's preferred in canopen:

### Style 1 - All in `__init__()`

```python
class Network(MutableMapping):
    """Representation of one CAN bus containing one or more nodes."""

    def __init__(self, bus: Optional[BusABC] = None):
        """
        :param can.BusABC bus:
            A python-can bus instance to re-use.
        """
        #: A python-can :class:`can.BusABC` instance which is set after
        #: :meth:`canopen.Network.connect` is called
        self.bus: Optional[BusABC] = bus
        #: A :class:`~canopen.network.NodeScanner` for detecting nodes
        self.scanner = NodeScanner(self)
        #: List of :class:`can.Listener` objects.
        #: Includes at least MessageListener.
        self.listeners: List[Listener] = [MessageListener(self)]
        self.notifier: Optional[Notifier] = None
        self.nodes: Dict[int, Union[RemoteNode, LocalNode]] = {}
        self.subscribers: Dict[int, List[Callback]] = {}
        self.send_lock = threading.Lock()
        self.sync = SyncProducer(self)
        self.time = TimeProducer(self)
        self.nmt = NmtMaster(0)
        self.nmt.network = self

        self.lss = LssMaster()
        self.lss.network = self
        self.subscribe(self.lss.LSS_RX_COBID, self.lss.on_message_received)
```

Pro: One location for defining the attributes and the type hints. Con: Can become messy with the type hints taking up the visual space.

### Style 2 - Keep type hints separate from `__init__()`

```python
class Network(MutableMapping):
    """Representation of one CAN bus containing one or more nodes."""

    bus: Optional[BusABC]
    scanner: NodeScanner
    listeners: List[Listener]
    notifier: Optional[Notifier]
    nodes: Dict[int, Union[RemoteNode, LocalNode]]
    subscribers: Dict[int, List[Callback]]
    send_lock: threading.Lock
    sync: SyncProducer
    time: TimeProducer
    nmt: NmtMaster
    lss: LssMaster

    def __init__(self, bus: Optional[BusABC] = None):
        """
        :param can.BusABC bus:
            A python-can bus instance to re-use.
        """
        #: A python-can :class:`can.BusABC` instance which is set after
        #: :meth:`canopen.Network.connect` is called
        self.bus = bus
        #: A :class:`~canopen.network.NodeScanner` for detecting nodes
        self.scanner = NodeScanner(self)
        #: List of :class:`can.Listener` objects.
        #: Includes at least MessageListener.
        self.listeners = [MessageListener(self)]
        self.notifier = None
        self.nodes = {}
        self.subscribers = {}
        self.send_lock = threading.Lock()
        self.sync = SyncProducer(self)
        self.time = TimeProducer(self)
        self.nmt = NmtMaster(0)
        self.nmt.network = self

        self.lss = LssMaster()
        self.lss.network = self
        self.subscribe(self.lss.LSS_RX_COBID, self.lss.on_message_received)
```

Pro: Create much cleaner `__init__()` which is easier to follow. Con: Need to keep type declarations in sync with implementation in `__init__()`.

acolomb, 2024-05-19:
I prefer style 1. The second version is semantically different, it declares a class attribute, later overridden by an instance attribute. That is IMHO unnecessary duplication and also wastes some memory because the class definition keeps its separate attribute although we're only interested in the instance attributes.

christiansandberg, 2024-05-19:
I don't have a strong preference but I usually do alternative 1, mostly because of less duplication and less risk for attributes and type hints to go out of sync.

sveinse, 2024-05-20:
While we're at it: What is the wanted convention for composite type definitions?

* Alternative 1:  `TODObject = Union[ODArray, ODRecord, ODVariable]`
* Alternative 2: `ODObjectT = Union[ODArray, ODRecord, ODVariable]`

Any other proposals?

acolomb, 2024-05-20:
No T?

sveinse, 2024-05-20:
> No T?

Hmm. How to differentiate between type annotation objects and regular objects that can be instantiated?

acolomb, 2024-05-20:
I've never seen Python code make the distinction using different names. Type hints are pretty flexible in use, whether it's an actual class or some type alias. All that matters is the type checker needs to make the right assumptions based on them. I don't think anyone would start writing code trying to instantiate a type alias, and not notice it's an error.

acolomb, 2024-06-11:
Inspired by the questions raised in #444, maybe the right place for such a question would be the "Discussions" feature on GitHub?  We seem to have a set of standard categories there, but nothing really fits development / coding style topics.

What do you think, should we get into a habit of moving issues with "user support" questions and "developer guidance" into some Discussions categories?

sveinse, 2024-06-11:
I think my question is answered, so let me be a good TS citizen and close this. Thanks for the reminder.

closed 2024-06-11 by sveinse