canopen-python/canopen #444: Remove old and stale branches from project?

friederschueler, 2024-05-27:
Just what the title says: I think we should remove old and stale branches from the project.

friederschueler, 2024-06-11:
If you like, you could give me some moderation rights and I could search the issues and close all the "support" one, which are questions about API usage. I will add an PR for automatically closing inactive issues.

acolomb, 2024-06-11:
I don't really like the idea of closing stuff automatically, but would rather look through the list manually.  Feel free to add comments where we don't expect any further progress, then I'll close them in a few days unless something new crops up.

Regarding the branches, I haven't looked through them but I'm sure some could be removed.  However, keeping them doesn't hurt and I would only remove those which have been merged anyway.  Nobody needs to fetch them if they don't want to.

acolomb, 2024-06-11:
I've started adding the `question` label to a few such issues, from a quick pass over the open issues list.  That should help in filtering out what we can close when not answered for a while.

@christiansandberg I can't set permissions on the repo, but you could probably give access to assign labels for @friederschueler and anyone else interested in working as issue triager?

friederschueler, 2024-06-12:
> @christiansandberg I can't set permissions on the repo, but you could probably give access to assign labels for @friederschueler and anyone else interested in working as issue triager?

I would like to triage issues. As I will be working at least 3+ more years with CanOpen, I can use some of my working time.

friederschueler, 2024-06-12:
Also I would like to introduce a "stale" and a "missing information" label.