canopen-python/canopen #445: Saving dynamic/empty PDOs: TypeError: unsupported operand type(s) for |: 'NoneType' and 'int'

friederschueler, 2024-05-27:
Currently, if you have dynamic PDOs, which are only defined but not configured in the EDS-File, you can not save them, because they are initialized with cob_id = None, which can not be compared in this loc in pdo/base.py:376

`self.com_record[1].raw = self.cob_id | PDO_NOT_VALID | (RTR_NOT_ALLOWED if not self.rtr_allowed else 0x0)`
     
Also, map is initilized with [], but the save-check checks for None, which is never the case, so all pdo will be saved, even if they are empty.

Test to reproduce:

```
    def test_save_pdo(self):
        node = canopen.Node(1, EDS_PATH)
        node.tpdo.save()
        node.rpdo.save()
```

acolomb, 2024-05-27:
This is a known problem with an existing workaround.  It's why we have the `.predefined_cob_id` attribute.  You can assign that to `.cob_id` before saving.

Nonetheless, making `pdo.save()` more robust against this doesn't hurt.  Will review your PR as time permits.

friederschueler, 2024-05-27:
predefined_cob_id will only be set to the first four PDOs.
My Workaround is to set cob_id to 0x0 for all PDOs for all devices and also set map to None to pass the check for writing.

The problem is that we initialize cob_id with None, but we need a int Value as soon as the check occurs, so this is bug. Also this is a very abstract error, which doesn't help the user to understand what happens.
And the wrong map check is even a bigger problem, sending hundreds of useless sdo-writes in a worst case scenario (which happens to occur to me). Wrote a comment on the PR, I can exlude the cosmetic stuff and log output.

PR #446 (2024-06-10): pdo: Skip saving if no COB-ID was set (fixes #445)
fixes #445

- changed save() Method of PdoBase to be more robust
- added some type hints
- removed unused variables
- added Test for PDO saving
- fixed long line (comments do count!)
commit fc577d1e (2024-06-10): pdo: Skip saving if no COB-ID was set (fixes #445) (#446)

The PdoBase.save() method would fail with a cryptic TypeError
exception when the COB-ID was not previously set (still at None).
Instead just skip the corresponding PDO and log an INFO level message
about it.  Add logging when re-enabling the PDO as well.

Also fix some minor issues in surrounding code:
* Added a test for PDO saving from EDS without further settings.
* Skip the always true condition "self.map is not None".  The
  attribute is initialized as an empty list.
* Simplify check for curtis_hack using getattr().
* Remove unnece [... cut]
closed 2024-06-10 by acolomb