canopen-python/canopen #449: Support CANopen (CiA 1301)?

sveinse, 2024-05-31:
Have it been considered to add support for CANopen FD (CiA 1301)? Is it of interest to add support for it in this library, or would it better live in another? 

The concepts are very similar, but there are protocol differences. E.g. EMCY have a larger payload, a new Universal SDO (USDO) protocol to mention two examples.

acolomb, 2025-02-16:
Someone with more experience / insight of CANopen FD needs to judge this.  I'd say if a lot of functionality from the current library can be re-used, then why not.  Certainly creating a separate library for FD will cause some extra effort that could be avoided.  And especially for users wanting to support both protocols in one application, the split would make it cumbersome I guess.

sveinse, 2025-06-22:
It would be nice to have CANopen-FD, but we're not overwhelmed by response for it, nor do we have much information about it. I'll close this issue for now, as we won't embark on FD without any additional push and input. We can reopen this issue when it does.

closed 2025-06-22 by sveinse