canopen-python/canopen #450: DCF Parsing: Lookup of baudrate incorrectly uses camel case (i.e, "BaudRate" instead of "Baudrate") which breaks loading of DCF

hepr-skylotec, 2024-06-03:
Loading of DCF looks up 'BaudRate' (camel case) instead of 'Baudrate'. This causes loading to fail if only 'Baudrate' is given in the DCF file.

The CiA 306, version 1.3.0 "CANopen electronic datasheet (EDS)" specification shows 'Baudrate' (without camelcase).

[image]

Suggested fix:
Change line 94 in objectdictionary/eds.py from "BaudRate" to "Baudrate"

acolomb, 2024-06-03:
There was a previous attempt to fix this: https://github.com/christiansandberg/canopen/pull/347

It somehow got mixed up with #350 and then stalled.  Ping @thesummer, will you be finishing either of those PRs?

hepr-skylotec, 2024-06-04:
@acolomb Thanks for the heads-up. I do not know why I did not find #347 .

I tracked this down to 074cbbc4d4c2c7803dc1e0151f6f534dbe44efc8 and reverted it for this specific change. I created a PR for it as #452

acolomb, 2024-06-04:
Indeed, that change looks pretty wrong.  @af-silva do you remember how you came to that conclusion?  The reference to #304 and #305 in the commit message doesn't really help.

af-silva, 2024-06-04:
@acolomb to be honest I don't recall but indeed is my bad, you are right! Following the CiA 306 Documentation Section 5.3.3.3

> Device Commissioning >> Baudrate shall indicate the device’s baudrate (Unsigned16)

For the DeviceCommissioning section the Baudrate is lower case while in the DeviceInfo section is camel case.

acolomb, 2024-06-04:
Also note that the spec contains a typo on "DeviceComissioning".  It should be spelled with two "m", as in the headline.  But that's how the format is specified, just need to be aware of the typo.