canopen-python/canopen #455: pyhton-can 4.4.0 breaks canopen 2.2.0

bjornolo, 2024-06-10:
CANopen automaticall gets the latest python-can 4.4.0 wich gives error:

TypeError: Can't instantiate abstract class MessageListener without an implementation for abstract method 'stop'

Reveting back to python-can 4.3.1 fixes issue

acolomb, 2024-06-10:
It's a known issue and already fixed on master. We just haven't got around to making a new release yet.

acolomb, 2024-06-10:
See #422.

acolomb, 2024-06-13:
Note that https://github.com/hardbyte/python-can/pull/1795 is already merged and will avoid this pitfall.  Dunno when that will land in a release on PyPI though, so we should still try to fix it soon on our end.

@christiansandberg I could make a release pretty soon, do I have your approval?  Not sure what needs to be done beyond the GitHub release though, will it land on PyPI automatically?

christiansandberg, 2024-06-13:
@acolomb go ahead and make a release if possible. It's nice to see so much activity but I haven't had the time to check all changes.

I hope it should automatically push to pypi but if not I can do it manually.

acolomb, 2024-06-13:
Alrighty, PyPI now has version 2.3.0, but Read The Docs hasn't picked it up yet.  Noted a small mistake just after the release: README still stated Python >=3.6 support, but that is now fixed on master at least.  The release notes do mention the upped dependency.

closed 2024-06-13 by acolomb