canopen-python/canopen #463: Online Documentation is (semi) broken

friederschueler, 2024-06-17:
https://canopen.readthedocs.io/en/latest/index.html

Every page is missing the API section (it is just empty) and there is also no link to github.

acolomb, 2024-06-17:
Confirmed, but right now I cannot even get local Sphinx invocation to work.  Will need to investigate where that goes wrong.  Apparently it has to do with `pkg_resources` no longer being available.

acolomb, 2024-06-18:
I hope #465 will fix the build on readthedocs.  #463 fixes some additional warnings during the Sphinx build.

acolomb, 2024-06-18:
Apparently the autodoc extension is not working on the build used by ReadTheDocs.  It does show content in my local build.

friederschueler, 2024-06-18:
I think only @christiansandberg can check this.

christiansandberg, 2024-06-18:
I see a lot of errors like this but haven't invested further yet.

```

WARNING: autodoc: failed to import class 'emcy.EmcyConsumer' from module 'canopen'; the following exception was raised:
Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/canopen/envs/latest/lib/python3.12/site-packages/sphinx/ext/autodoc/importer.py", line 143, in import_module
    return importlib.import_module(modname)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docs/.asdf/installs/python/3.12.0/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1381, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1354, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1325, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 929, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 994, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/home/docs/checkouts/readthedocs.org/user_builds/canopen/checkouts/latest/canopen/__init__.py", line 1, in <module>
    from canopen.network import Network, NodeScanner
  File "/home/docs/checkouts/readthedocs.org/user_builds/canopen/checkouts/latest/canopen/network.py", line 19, in <module>
    from canopen.node import RemoteNode, LocalNode
  File "/home/docs/checkouts/readthedocs.org/user_builds/canopen/checkouts/latest/canopen/node/__init__.py", line 1, in <module>
    from canopen.node.remote import RemoteNode
  File "/home/docs/checkouts/readthedocs.org/user_builds/canopen/checkouts/latest/canopen/node/remote.py", line 4, in <module>
    from canopen.sdo import SdoClient, SdoCommunicationError, SdoAbortedError
  File "/home/docs/checkouts/readthedocs.org/user_builds/canopen/checkouts/latest/canopen/sdo/__init__.py", line 1, in <module>
    from canopen.sdo.base import SdoVariable, SdoRecord, SdoArray
  File "/home/docs/checkouts/readthedocs.org/user_builds/canopen/checkouts/latest/canopen/sdo/base.py", line 7, in <module>
    from canopen import objectdictionary
  File "/home/docs/checkouts/readthedocs.org/user_builds/canopen/checkouts/latest/canopen/objectdictionary/__init__.py", line 281, in <module>
    class ODVariable:
  File "/home/docs/checkouts/readthedocs.org/user_builds/canopen/checkouts/latest/canopen/objectdictionary/__init__.py", line 290, in ODVariable
    INTEGER24: IntegerN(24),
               ^^^^^^^^^^^^
TypeError: Struct() argument 1 must be a str or bytes object, not int
```

acolomb, 2024-06-18:
Thanks for checking.  Wow, that makes absolutely no sense.  The `IntegerN` class is imported via `from canopen.objectdictionary.datatypes import *`, and it *derives* from `Struct`, but we are not *instantiating* a `Struct`.  The parent class constructor is called with a string argument for sure.  The only thing I can think of is that Sphinx somehow messes up the asterisk import, thus we could try with an explicit name import.

Any other ideas?

acolomb, 2024-06-18:
Oh, that explicit named import seems to have fixed it?!  At least I do see an API section again now on the "latest" site.

acolomb, 2024-06-18:
@christiansandberg Would you mind allowing me to access / administrate the Read the Docs project as well?

christiansandberg, 2024-06-18:
Yes very strange. Invitation sent.

commit 8f89d428 (2024-06-18): doc: Clean up and modernize Sphinx configuration (fixes #463) (#465)

* Switch from setuptools' pkg_resources to importlib.metadata.

The remaining changes are based on a fresh invocation of sphinx-quickstart (version 7.2.6), trying to format our existing modifications in the same style.

* Update string literals to assume UTF-8 default.
* Remove commented Sphinx configuration options.
* Remove Sphinx options which are at their default.
* Reorder, project settings first.
* Remove explanatory comments on configuration options.
* Include section headers as generated via sphinx-quickstart.
* Remo [... cut]
closed 2024-06-18 by acolomb